pyhys/minimalmodbus #1: close_port_after_each_call logic doesn't take exceptions into account

photron, 2015-10-29:
The _communicate function assumes that the serial port is closed if close_port_after_each_call is set and tries to open it. But If it is already open the serial.open call will fail with an exception saying that it is already open.

This situation can be triggered if any unhandled exception occurs between the serial.open call and the serial.close call in the _communicate function. There are several places that can raise such exceptions:

* the serial.write call can fail with an SerialTimeoutException
* the local echo logic can raise an IOError
* I think in my case I also saw the serial.read call fail, even if the pySerial documentation doesn't say it could

Probably the easiest fix is to put all code in the _communicate function between the serial.open call and the serial.close call into a try/finally block and move the serial.close call itself in the finally part of it.

ticino, 2016-05-02:
I run in this issue too: When a second process (multiprocessing package) tries to read from an already "opened" device (due to bad timing). 
Fixed it in my own code by introducing an additional lock, for now. But it would be nice to handle such a case already here ...

pyhys, 2019-08-03:
Next release will check if the port is open, and only open it if necessary.

closed 2019-08-03 by pyhys