pyhys/minimalmodbus #3: The slave is indicating an error

kxmatejka, 2015-11-06:
Hello, I have electricity meter Schneider iem 3155 (http://www.schneider-electric.com/en/product-range/61273-acti-9-iem3000-series/)

I'm loading without any problem using your library registers: 3204, 3208, 3220, 3224, 3000, 3002, 3004, 3010

But now I need run command number 2008, action = write, size = 1, type = Uint16, possible values = 1-4

When I use
instrument.write_long(2008, 1)

I get back
ValueError: The slave is indicating an error. The response is: '\x02\x90\x04\xbd\xc3'

I use python 2.7

Shneider support told me set on e. meter Com. Protection like disabled, I did, and no change.
You're my last resort. Some idea?

photron, 2015-11-09:
The write_long function takes an 32-bit integer value, splits it into to 16-bit values and writes them to two registers. In your case it'll try to write to register 2008 and 2009.

But according to

> action = write, size = 1, type = Uint16, possible values = 1-4

it looks to me that you should only write to a single register. My guess is that the error might come from trying to write to register 2009.

Try using write_register instead of write_long to only write one 16-bit integer to register 2008 and not touch register 2009:

```
instrument.write_register(2008, 1)
```

kxmatejka, 2015-11-09:
thank you for your wise advice, but result is still same

full output
```py
Traceback (most recent call last):
  File "./prepnuti-tarifu.py", line 18, in <module>
    instrument.write_register(2008, 1)
  File "/usr/local/lib/python2.7/dist-packages/minimalmodbus.py", line 292, in write_register
    self._genericCommand(functioncode, registeraddress, value, numberOfDecimals, signed=signed)
  File "/usr/local/lib/python2.7/dist-packages/minimalmodbus.py", line 693, in _genericCommand
    payloadFromSlave = self._performCommand(functioncode, payloadToSlave)
  File "/usr/local/lib/python2.7/dist-packages/minimalmodbus.py", line 794, in _performCommand
    payloadFromSlave = _extractPayload(response, self.address, self.mode, functioncode)
  File "/usr/local/lib/python2.7/dist-packages/minimalmodbus.py", line 1067, in _extractPayload
    raise ValueError('The slave is indicating an error. The response is: {!r}'.format(response))
ValueError: The slave is indicating an error. The response is: '\x02\x90\x04\xbd\xc3'
```

I got the feeling that I'm writing to the register incorrectly from beginning

here is table from documentation for setting tariffs from which I deduced
https://drive.google.com/file/d/0B3I_MjM8ksB2OTEtYTFSWDJBMDg/view
```py
instrument.write_register(2008, 1)
```

and here is table for setting date and time
https://drive.google.com/file/d/0B3I_MjM8ksB2LUpwc21Lb2hDcG8/view

1 command number and 6 possible values. Do you understand me? How to recognize the units? Command for setting the tariff is probably wrong.

photron, 2015-11-09:
The response you're getting indicates Modbus exception code 4. Here's the Modbus specification about exception code 4:

> 04 | SERVER DEVICE FAILURE | An unrecoverable error occurred while the server was attempting to perform the requested action.

According to the documentation you linked in:

> NOTE: You can only set the tariff using this method if the Tariff Mode is set to by Communication.

Maybe the Tariff Mode is not set to "by Communication" and the device raises exception code 4 because you're trying to set the tariff anyway.

Maybe you need to set register 2060 to 1 before you can set register 2008 to 1?

Yes the data/time thing is a bit strange. Maybe it's supposed to work like this: the command number defines the base register number, 1003 in case of the date/time command. The command takes 8 UInt16 values in 8 registers stating from 1003. So you could use

```
instrument.write_registers(1003, [0, 2015, 10, 9, 12, 56, 11, 0])
```

to set the date and time to November 9, 2015 at 12:56:11.

Or maybe if you're not allowed to write to the reserved parts:

```
instrument.write_registers(1004, [2015, 10, 9, 12, 56, 11])
```

If it works that way then you need to do

```
instrument.write_registers(2060, [0, 1]) # set Tariff Mode to "by Communication"
instrument.write_registers(2008, [0, 1]) # set tariff to T1
```

to set the tariff to T1. Or if you're not allowed to write to the reserved parts

```
instrument.write_registers(2061, [1]) # set Tariff Mode to "by Communication"
instrument.write_registers(2009, [1]) # set Tariff to T1
```

But I'm just guessing here based on the documentation you linked in. You'll have to try to see what works.

kxmatejka, 2015-11-09:
good tips, this I set on e. meter
> NOTE: You can only set the tariff using this method if the Tariff Mode is set to by Communication.

Each command returns 04, but I have idea, to enter the settings you need to enter a password. For 100 % must be authorized despite Modbus.

Once again will try to contact support.

kxmatejka, 2015-11-23:
Hello, I have new information.
I need write into register 5250 and the command number 2008 must be probably part of data.
[image]
any idea how to write this?
this return None
```py
instrument.write_registers(5250, [2008, 3]) # last parameter can by 1-4
```
but this isn't correct, data from the tariff 3 are still 0.

photron, 2015-11-23:
Okay, it works differently from what I was guessing about it.

You're almost there. Your example call writes 2008 to register 5250 and 3 to register 5251:

```
instrument.write_registers(5250, [2008, 3])
```

But according to the documentation, the 3 needs to be written to register 5252.

To match what the box with the red border shows, you need to call this:

```
instrument.write_registers(5250, [2008, 0, 3])
```

The 0 is in there to go into register 5251, that is marked as reserved.

kxmatejka, 2015-11-23:
interesting, so this can be also written
```py
instrument.write_registers(5250, [2008, 0, 3])
```
like this?
```py
instrument.write_register(5250, 2008)
instrument.write_register(5252, 4)
```

but still not working, I also reading register 5376 and then always return code 3000 (= invalid command)

photron, 2015-11-23:
If you split it in two calls, then the device will receive two Modbus write requests. According to the documentation you linked it, a command has to be received as a single Modbus write request. So you cannot split it.

Did you enable "Tariff Mode to by Communication"? I think you need to call

```python
instrument.write_registers(5250, [2060, 0, 1])
```

before you can call

```python
instrument.write_registers(5250, [2008, 0, 3])
```

kxmatejka, 2015-11-23:
Thank you so much, this I thought was irrelevant. Now tariff switching finally works! Lets go try set date.

closed 2015-11-23 by kxmatejka