pyhys/minimalmodbus #6: Not getting response from a system I am trying to use Minimal Modbus to talk to

nbyrne, 2016-04-14:
>>> print instrument.read_register(0, 1)

MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x01\x03\x00\x00\x00\x01\x84\n' (01 03 00 00 00 01 84 0A)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 1310283.0 ms, minimum silent period: 8.02 ms.
MinimalModbus debug mode. Response from instrument: '' () (0 bytes), roundtrip time: 51.0 ms. Timeout setting: 50.0 ms.

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "minimalmodbus.py", line 258, in read_register
    return self._genericCommand(functioncode, registeraddress, numberOfDecimals=numberOfDecimals, signed=signed)
  File "minimalmodbus.py", line 697, in _genericCommand
    payloadFromSlave = self._performCommand(functioncode, payloadToSlave)
  File "minimalmodbus.py", line 795, in _performCommand
    response = self._communicate(request, number_of_bytes_to_read)
  File "minimalmodbus.py", line 930, in _communicate
    raise IOError('No communication with the instrument (no answer)')
IOError: No communication with the instrument (no answer)

nbyrne, 2016-04-14:
I see same issue in Windows 7 (shown above) and in Ubuntu.

I am using a Null Modem cable to talk to a CSK EZT 570i environmental test chamber

nbyrne, 2016-04-14:
If I even unplug the null modem cable from chamber I get same response.
Have tried different baud rates. 
Tried different timeouts: 200ms 600ms 800ms
Chamber does not want to talk back to me.

pyhys, 2016-04-14:
Hej,
I did find the manual at
http://www.cszindustrial.com/portals/1/pdf/ezt_570i_user_manual_reva.pdf
and
http://www.cszindustrial.com/portals/1/pdf/ezt_570i_user_communication_r
eference_manual_reva.pdf

"The EZT-570i's communication address, parity and timeout can be changed
on the Web Server/Modbus/VNC screen located under
the chamber setup menu." Make sure this match the settings in
MinimalModbus, as well as the chamber Modbus address (unclear how to set
it).

Try reading the temperature (the example given in the documentation):
instrument.read_register(61, 1) 

Verify that MinimalModbus sends:
01 03 00 3D 00 01 15 C6

Then there should be some response. Otherwise see the MinimalModbus
section on communication troubleshooting:
https://minimalmodbus.readthedocs.org/en/master/troubleshooting.html
<http://https://minimalmodbus.readthedocs.org/en/master/troubleshooting.
html> 

Best regards
Jonas

<-----Ursprungligt Meddelande----->

am trying to use Minimal Modbus to talk to (#6) 

<https://github.com/notifications/beacon/ABg1jpPfdZg84aeejwDS5_thQCq5Hci
Kks5p3o2lgaJpZM4IHqqA.gif>

nbyrne, 2016-04-14:
Thanks Jonas

This is what I get:

When I try read temperature of chamber:

>>> print instrument.read_register(61, 1)

what I send:
MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x01\x03\x00=\x00\x01\x15\xc6' (01 03 00 3D 00 01 15 C6)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 86944505.0 ms, minimum silent period: 8.02 ms.

(01 03 00 3D 00 01 15 C6) 
CRC bytes: 15 C6 
matches the expected bytes CSZ manual listed for this register to read temperature of chamber using Modbus protocol: all the bytes match what manual expects us to send when reading temperature.

So this means the CRC format is correct when I am attempting to read temperature.
However I simply just do not get response back from chamber.
So maybe issue with timing/baud rate or the cable or even the controller is just not recognizing what I am sending.
I have ordered Nullmodem-RS232 adapter and waiting on it to see if it helps.

I also contacted CSZ about this and sent them details on what I see when attempting read
temperature register. (listed above)

thanks
Niall

nbyrne, 2016-04-14:
[image]

nbyrne, 2016-04-14:
so what the MinimalModbus command sends for bytes appears to match what their manual expects me to send for bytes to read back temperature: 01 03 00 3D 00 01 15 C6

nbyrne, 2016-04-15:
[image]

nbyrne, 2016-04-15:
That waveform is TX output of null modem cable RS-232. I am checking it is correct sequence to match the bytes above (trying to read temperature register 61 from the CSZ)
But it appears my hardware and cable is attempting to communicate with the chamber.
I have yet to capture on scope if chamber responding to this sequence would need connect scope to the TX pin on chambers RS-232 connector,  will attempt this probably Monday but requires some possible jerry rigging to get a scope probe on there.

nbyrne, 2016-04-18:
We have a solution. The CSZ manual specified to use a null-modem cable (cross-over) RS-232 connection.

nbyrne, 2016-04-18:
[image]

nbyrne, 2016-04-18:
This is incorrect. 
We need use a straight-through RS-232 cable to communicate with chamber.
Now I am sending data correctly from straight through cable's TXD pin 2 to chamber's RXD pin 2 using straight-through cable.
Using null modem cable cross over cable sends the data to Pin 3 the chambers TXD pin, which is  incorrect. This was the problem.

I believe their document needs be updated so sent them details on this.

nbyrne, 2016-04-18:
[image]
Now MinimalModbus works !

closed 2016-04-20 by nbyrne