pyhys/minimalmodbus #11: several reads fails on 0.7 with "IOError: No communication with the instrument (no answer)" but not on 0.6

KennethNielsen, 2016-10-18:
Hallo minimalmodebus devs

I have an issue with the latest version. I have a drivers that runs perfectly with version 0.6, but with 0.7 it fails with the error mentioned in the subject. The debug information does not tell me very much (except that it times out):

```text
MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x02\x03\x00y\x00\x01U\xe0' (02 03 00 79 00 01 55 E0)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 20.1 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '' () (0 bytes), roundtrip time: 2003.0 ms. Timeout setting: 2000.0 ms.
```
I guess my best bet is that something changed between 0.6 and 0.7 that changes how the lower level communication via serial is done.

The program is running on raspbian jessie. Serial version i 3.2.1.

I am happy to gather additional information as required.

pyhys, 2016-11-01:
> No communication with the instrument (no answer)" but not on 0.6 (#11)

Best Kenneth!
Please try both with version 0.6 and 0.7, and verify that the sent data
is "02 03 00 79 00 01 55 E0" in both cases. MinimalModbus just builds
this data, and is interpreting what is sent back. If the sent data is
same for both versions, then maybe there is something wrong with the
pyserial library (or I use it in wrong way).

Best regards
Jonas

<-----Ursprungligt Meddelande----->

6PItR7_xfcnks5q1IcZgaJpZM4KZhjZ> .
<https://github.com/notifications/beacon/ABg1jrkznVsXwKWm2yA8L2q9Z6p4xDb
8ks5q1IcZgaJpZM4KZhjZ.gif>

KennethNielsen, 2016-11-23:
Hallo Jonas

I finally found some time to investigate. As it turns out it happens only with Python3. I have a acquired debug information here:

```text
MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x02\x03\x00m\x00\x01\x15ä' (02 03 00 6D 00 01 15 E4)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 21.1 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '\x02\x03\x02ÿÿýô' (02 03 02 FF FF FD F4) (7 bytes), roundtrip time: 82.6 ms. Timeout setting: 2000.0 ms.

MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x02\x03\x00m\x00\x01\x15ä' (02 03 00 6D 00 01 15 E4)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 20.9 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '\x02\x03\x02ÿÿýô' (02 03 02 FF FF FD F4) (7 bytes), roundtrip time: 82.4 ms. Timeout setting: 2000.0 ms.

MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x02\x03\x00m\x00\x01\x15ä' (02 03 00 6D 00 01 15 E4)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 20.7 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '' () (0 bytes), roundtrip time: 2003.2 ms. Timeout setting: 2000.0 ms.
```

as it can be seen, it sends the exact same bytes, so you are right it is probably not a minimalmodbus issue. It does however use the same serial version, so I'm really confused.

Doing the same thing with minimalmodbus 0.6 gives no error and this output:

```text
MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x02\x03\x00m\x00\x01\x15ä'
MinimalModbus debug mode. No sleep required before write. Time since previous read: 83.1 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '\x02\x03\x02ÿÿýô' (7 bytes), roundtrip time: 91.2 ms. Timeout setting: 2000.0 ms.

MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x02\x03\x00m\x00\x01\x15ä'
MinimalModbus debug mode. No sleep required before write. Time since previous read: 86.0 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '\x02\x03\x02ÿÿýô' (7 bytes), roundtrip time: 88.6 ms. Timeout setting: 2000.0 ms.

MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x02\x03\x00m\x00\x01\x15ä'
MinimalModbus debug mode. No sleep required before write. Time since previous read: 81.5 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '\x02\x03\x02ÿÿýô' (7 bytes), roundtrip time: 94.4 ms. Timeout setting: 2000.0 ms.
```

one thing that can of course be seen is that 0.7 is a lot faster. So maybe the explanation is as simple as the speedup from 0.6 to 0.7 (I believe it was a faster hashing algorithm) means that communication now moves too fast for the equipment. So if I just slow it down a little somehow it may be ok. I will experiment with that later.

KennethNielsen, 2017-03-23:
I'll close this as not a minimalmodbus issue

closed 2017-03-23 by KennethNielsen