pyhys/minimalmodbus #12: Feature request: custom minimum silent period.

grinsted, 2017-03-14:
My instrument does not respond reliably unless i add a tiny sleep before I read from a register. It would be awesome if it would be possible to configure the minimum silent period (e.g. to 4chars instead of 3.5).

rb1205, 2017-03-21:
I have some problem too with the default setting on sequential reads using a fast computer with no running app except for the modbus client. Curiously enough the IO is fine if I have a CPU load like a VNC server. OS is windows 7, 9600 baud, communicating with Eurotherm 2400 devices.

Setting MINIMUM_SILENT_CHARACTERTIMES in minimalmodbus.py to 5 solved the issue for me.

pyhys, 2019-07-27:
The minimum silent period will be increased for high baudrates in next release, see #25.

closed 2019-07-27 by pyhys