pyhys/minimalmodbus #14: read_register() and write_register() failing

aarontusko, 2017-05-31:
When I call read_string( 19, 7 ) I get a response as expected. But when I call write_register( 99, 0 ) I get an error. Then when I call read_string( 19, 7) again I get an error (where it succeeded before), but all calls after that are fine. 

This will happen with either read_register() or write_register() where they will throw an exception and print an incorrect response packet, and complain about the CRC.

My modbus instrument seems to be working fine. When I write_register() I can see the outputs change as I commanded. So I know the instrument is receiving the command. I also hooked up a second RS485toUSB converter (on a seperate computer) up so I could snoop on the transaction. Both the query and the response look fine on the bus, but the response is garbled in the exception report.  And as said before the modbus instrument is responding correctly.

Here are some scripts that I wrote that demonstrate the issue.
minimal_bug.py  (This demonstrates the issue)
minimal_bug_OUTPUT.txt  (Annotated output from minimal_bug.py)
hexdump.py  (snooping on the RS485 line from my linux box)
hexdump_OUTPUT.txt  (Annotated output from hexdump.py, shows the correct query and response)

System running minimal_bug.py is:
Windows 10
python 2.7
minimalmodbus 0.7
minimalmodbus.serial 3.3

Please excuse my poor python skills, I am still new to it.

[minimalmodbus_bug_scripts.zip](https://github.com/pyhys/minimalmodbus/files/1042789/minimalmodbus_bug_scripts.zip)

rdpoor, 2017-11-13:
> Both the query and the response look fine on the bus, but the response is garbled in the exception report. 

I agree.  I've re-indented your hexdump bus snoop to show the query and response better:

    Query				Response
    0A 03 00 13 00 07 F4 B6           0A 03 0E 02 00 02 00 02 00 02 00 02 00 02 00 02 00 43 0B 
    0A 03 00 13 00 07 F4 B6           0A 03 0E 02 00 02 00 02 00 02 00 02 00 02 00 02 00 43 0B 
    0A 03 00 13 00 07 F4 B6           0A 03 0E 02 00 02 00 02 00 02 00 02 00 02 00 02 00 43 0B 
    0A 10 00 63 00 01 02 00 00 DC F3  0A 10 00 63 00 01 F0 AC 
    0A 10 00 63 00 01 02 03 FF 9C 43  0A 10 00 63 00 01 F0 AC 
    0A 10 00 63 00 01 02 00 00 DC F3  0A 10 00 63 00 01 F0 AC 
    0A 03 00 13 00 07 F4 B6           0A 03 0E 02 00 02 00 02 00 02 00 02 00 02 00 02 00 43 0B 

I've not manually verified the checksums, but following request number 4 (write multiple registers), minimalmodbus seems to be getting the response from request number 3 (read register) of `0A 03 0E 02 00 02 00 02` rather than `0A 10 00 63 00 01 F0 AC`:

    instrument.write_register( 99, 0x0000 )
    Checksum error in rtu mode: '\x00\x02' instead of 'fX' . The response is: 
    '\n\x03\x0e\x02\x00\x02\x00\x02' (plain response: '\n\x03\x0e\x02\x00\x02\x00\x02')

pyhys, 2019-07-27:
This is probably the same as the problem targeted by #33

closed 2019-07-27 by pyhys