pyhys/minimalmodbus #15: Problem with IOError line 930

krismc, 2017-07-21:
I poll 3 registers with 3 uses of the raw=slave.read_register(x) cmd on a RPi running python 2.7 under Jessie Lite.  I've noticed if I get a legitimate I/O error on the first register then I get bogus I/O errors for the other two. The USB to RS232 dongle does not flash it's Tx light for the next 2 reads, so I'm getting those I/O errors because minimal modbus is not sending for the other two. Somehow the 1st error does something inside minimal modbus that prevents it sending for the next 2 slave.read cmds.
Bizarrely, if I close & rerun my program all is good again till the next legitimate IOError then complete lockout again for any subsequent reads, again because minimal modbus is not sending. 
Upon further checking I've realised this also applies to any other errors raised by minimal modbus. For example the ValueError (when response checksum is wrong) triggers indefinite IOErrors from then on, again because minimal modbus is not sending. It says it's sending (with diagnostic mode on) but it isn't.

llamicron, 2017-08-12:
Same issue here

pyhys, 2019-07-27:
This is probably the same as the problem targeted by #33

closed 2019-07-27 by pyhys