pyhys/minimalmodbus #16: Error, if ASCII_HEADER on write is different from response

markmuzenhardt, 2017-07-26:
I already fixed that by implementing another sys-parameter named _ASCII_RESP_HEADER which by default is set to the same value like _ASCII_HEADER. This enables the user to change both headers separately, which I needed on my last project.

[minimalmodbus.txt](https://github.com/pyhys/minimalmodbus/files/1176578/minimalmodbus.txt)

pyhys, 2019-07-27:
This seems to be a deviation from the MODBUS ASCII standard. If there is a common need for this, please reopen.

closed 2019-07-27 by pyhys

Angus71, 2023-05-06:
Hi,

I stumbled upon a set of devices (ABL Wallboxes [Protcol description](https://www.ablmobility.de/global/downloads/anleitungen/emh1/Schnittstellenbeschreibung_Modbus-ASCII.pdf?m=1652185246&)), which also use a non standard response header in their MODBUS ASCII protocol.

As I didn't find a generic solution, I added a property to the Instrument class, which can be set after instanciating (Defaults to _ASCII_HEADER).

If required / wanted, I can provide a pull request for review and later integration. Just let me know, if this feature is still accepted, as it violates the MODBUS ASCII standard and in the last 4 years nobody actually commented on this :smiley: