pyhys/minimalmodbus #17: Package puts some inappropriate modules into root namespace

bilderbuchi, 2017-07-31:
Hi!
Thank you for this package, and especially the very detailed documentation!

One surprising thing that I found is that this package puts `dummy_serial`, `eurotherm3500` and `omegacn7500` modules into the "root" namespace of installed python packages. I found this weird and surprising/unexpected. Let me show you:
```
Jupyter QtConsole 4.3.0
Python 3.6.2 |Anaconda custom (64-bit)| (default, Jul 20 2017, 12:30:02) [MSC v.1900 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 6.1.0 -- An enhanced Interactive Python. Type '?' for help.

import numpy as np  # as expected

np
Out[2]: <module 'numpy' from 'C:\\Continuum\\Anaconda3\\lib\\site-packages\\numpy\\__init__.py'>

import minimalmodbus  # as expected

minimalmodbus
Out[4]: <module 'minimalmodbus' from 'C:\\Continuum\\Anaconda3\\lib\\site-packages\\minimalmodbus.py'>

import eurotherm3500

# What did I just import? What is this?

eurotherm3500
Out[7]: <module 'eurotherm3500' from 'C:\\Continuum\\Anaconda3\\lib\\site-packages\\eurotherm3500.py'>

# hmm, no idea from which package that came from...

import dummy_serial

dummy_serial
Out[10]: <module 'dummy_serial' from 'C:\\Continuum\\Anaconda3\\lib\\site-packages\\dummy_serial.py'>

#same deal here

minimalmodbus.__version__  # for completeness
Out[12]: '0.7'
```

I think it would be preferable to have these obviously minimalmodbus-related modules remain within the minimalmodbus namespace, by refactoring into a package and adding these as modules. I think even the normal usage API can be preserved by some `from minimalmodbus import *` or similar within the `__init__py`.

pyhys, 2019-07-29:
At least the eurotherm3500 and omegacn7500 modules will be removed from next release.

closed 2019-07-29 by pyhys

bilderbuchi, 2019-07-29:
👍 thanks! Maybe in the meantime, there is functionality available, e.g. from pyserial, to use as a dummy serial.