pyhys/minimalmodbus #20: read the bit status of internal relay of plc

vladislav-konoplyev, 2017-10-05:
Hello! I have Fatek PLC FBs-32MN and need to get bit status of internal relay.
I try to use read_bit() with function code = 2, it's return error(wrong adress). 

Which function should i use? and which function code?

here some doc with memory allocation of this PLC:
[doc](http://www.fatek.com/en/data%2Fftp%2FPLC%2FFBs_Manual%2FManual_1%2FInstruction%2FChapter_2.pdf)

P.S. i try to read registers and output relays and its works

andreyfesh, 2017-11-28:
which relay status are you trying to read?

vladislav-konoplyev, 2017-11-28:
in discrete status area(4096) -> M(2002). i can't read M's

andreyfesh, 2017-11-29:
I found similar situation in one of the plc's that I use. First of all according to your document, M2002 doesn't exist. If you look at your discrete status area memory allocation table you will see that it starts at X(256), Y(256), T(256), C(256), TR(40), S(1000) and then M(2002). That means that first register 0000 is X000.
Alternatively M block registers would be addressed as: 
X(256) + Y(256) + T(256) + C(256) + TR(40) + S(1000) - 1(index starts at 0) = 2063
From that we can assume that first M0 relay is at address 2064.
Try read_bit(2064, 1) to read status of M0.
In terms of function codes:
1 - READ discreet output coils
2 - READ discreet input contacts
3 - READ analog holding registers
4 - READ analog input registers

vladislav-konoplyev, 2017-11-29:
Thank you very much! I'll try soon. And then tell you about this experience

pyhys, 2019-07-27:
This seems to be solved

closed 2019-07-27 by pyhys