pyhys/minimalmodbus #22: Handling multiple masters

rdpoor, 2017-11-13:
Imagine a case where I have more than one USB <=> RS485 Modbus dongles on one machine.  And further imagine that each RS485 Modbus has more than one slave attached to it.  (In fact, in our testing environment, that's exactly what we're facing.)

Unless I'm mistaken, minimalmodbus is not set up to handle multiple masters nor multiple slaves on one bus.  I see two problems in the current structure:
* Communication parameters are all class variables (e.g. `minimalmodbus.BAUDRATE`) and cannot be set per master.
* Port names (e.g. `COM4` or `/dev/cu.usb01`) are associated with the Instrument and not with the master.

So, some release in the future might consider a refactoring to make this kind of thing possible:

    masterA = minimalmodbus.Master('COM4')
    masterA.baud_rate = 19200
    masterA.timeout = 0.01
    masterB = minimalmodbus.Master('COM5')
    masterB.baud_rate = 38400

    instrumentA1 = minimalmodbus.Instrument(masterA, 1)  # modbus address 1
    instrumentA2 = minimalmodbus.Instrument(masterA, 2) # modbus address 2
    instrumentB1 = minimalmodbus.Instrument(masterB, 1) # modbus address 1

I'd welcome thoughts and discussion on whether or not this is a good direction.

rdpoor, 2017-11-17:
Looking over the code, with a little cleverness, I think this should be possible to do and maintain full backward compatibility.  When I finish current obligations I'll look into doing a fork and pull request.

glennra, 2018-03-12:
I was about to ask exactly this question and found this issue. It would useful to get some input from the original author(s) about this.

pyhys, 2019-07-27:
It is possible to use several instruments on the same bus, as they share the same "serial" object instance. Do not change `minimalmodbus.BAUDRATE` (it is the default baud rate, a better name would have been DEFAULT_BAUDRATE).

Instead use: 
`instrumentA.serial.baudrate = 19200`

If instrumentB uses the same serial port, also its baudrate will be changed.

The documentation on this will be improved in next release.

closed 2019-07-27 by pyhys

rdpoor, 2019-07-27:
Jonas: 

Thank you for the clarification.  I realized I made an error by wrapping two separate issues in one bug report.  

If I understand your response correctly, it's still not possible to have two modbus masters (i.e. two USB interfaces) controlled by minimalmodbus -- do you agree?  If so, I can file a separate issue.

Thanks.

pyhys, 2019-07-28:
Yes, minimalmodbus is supposed to handle several instruments per serial port, and several serial ports. I agree that the documentation needs to be clarified. Please do some experiments with this, and come back if you find any problems.

Updated documentation:

`Use a single script for talking to all your instruments (if connected via the
same serial port). Create several instrument objects like::

    instrumentA = minimalmodbus.Instrument('/dev/ttyUSB1', 1)
    instrumentA.serial.baudrate = 9600
    instrumentA.serial.timeout = 0.2
    instrumentA.mode = minimalmodbus.MODE_RTU

    instrumentB = minimalmodbus.Instrument('/dev/ttyUSB1', 2)
    instrumentB.mode = minimalmodbus.MODE_ASCII

    instrumentC = minimalmodbus.Instrument('/dev/ttyUSB2', 1)

The instruments sharing the same serial port has the same ``serial`` Python object, so 
instrumentB will have the same baudrate and timeout as instrumentA.

You can use instruments on different serial ports in the same script, but
running several scripts using the same port will give problems.`