pyhys/minimalmodbus #24: Negative temperatures when reading from Eurotherm3508

ColinWaddell, 2017-12-04:
Morning.

I was using the ```eurotherm3500.py``` over the weekend to record temperatures from our oven as it was running an experiment.

Noticed in the logged data that negative numbers did not look like they were being converted correctly to a signed integer. For example the value -0.1 was being recorded as 6553.5

I added a small kludge to my code which sorted everything out:

    def eurotherm_sanitize_temp(temp):
        raw = int(temp*10)
        if raw > 32767:
            temp = -0.1*float(65536 - raw)
        return temp

Hope this note is of some use to you. You saved me a tonne of work by making this code available, so many thanks!

pyhys, 2019-07-29:
Hi, and thanks for your report.
I guess the way to correct this is to replace:
`self.read_register(289, 1)`
with:
`self.read_register(289, 1, signed=True)`

Unfortunately I do not longer have access to any Eurotherm instruments, so I need to drop support for them in next release of MinimalModbus.

closed 2019-07-29 by pyhys

ColinWaddell, 2019-07-29:
Ahh, that's a shame. We're using this library to control thermal-ovens and vacuum chambers for testing satellites. It's been mega useful for testing weird environmental profiles.

I understand though how dangerous it is to blindly write software for unknown hardware though and appreciate the heads-up about discontinued support.

pyhys, 2019-08-08:
Feel free to start a new project utilizing the old code 😄