pyhys/minimalmodbus #26: Choose CRC algorithm

Joe-Beach, 2017-12-04:
We want to use minimalmodbus to talk to a Sierra Compod interface on some lab equipment. We have determined that the unit uses a CRC-16 CCITT algorithm, which is different than the CRC-16 algorithm that minimalmodbus uses. 

Browsing the code, it looks like switching algorithms should be a matter of changing the lookup table and possibly a function definition. We have two questions: 

1. Is there anyone here who could walk us through modifying the code to calculate the CRC's?
2. Would you consider an option to choose the CRC algorithm in later releases?

Thanks!

mradziwo, 2018-06-09:
minimalmodbus is an implementaiotn of modbus standard which asserts particular type of CRC is used. So there is no need for changing it or allowing to use any other - however for your custom aaplication it may be wort a try to get this functionality by inheritance.
Are you still in need fr solution?

Joe-Beach, 2018-06-11:
Hello,

We actually figured the problem out, so we don't need the different CRC. 
The documentation for the device had some confusing examples that 
implied a nonstandard CRC, but the actual device followed the standard. 
We have been using minimalmodbus for all of our communication with it, 
thank you. Thank you for following up, too.

Joe

mradziwo, 2018-06-11:
No problem - I was in need to tune minimalmodbus - due to some sloppy
implementation o Modbus slave in device I suppose to communicate with.
So in case you would be in need to tune CRC, frame format, use other
function codes or even custom functions - feel free to may tap my shoulder.

Best regards,
Michał Radziwon

pyhys, 2019-07-27:
Nice that the problem is solved.

closed 2019-07-27 by pyhys