pyhys/minimalmodbus #27: How to write multiple coils

MarcoCovarrubias, 2018-03-09:
Hi, sorry for the silly question. 

Is just that I do not know how to write multiple coils at a time. The provided function (write_bit) throws an error when I try to write a value grater than 1. Forcing the argument functioncode = 15 in the fucntion call doesn't work either. Looking at the source code apparently there is no obvious implementation of this function. What am I missing?

Thanks in advance 👍

Devangsavla, 2018-08-23:
Exactly my doubt. Even the documentation has no mention of any function for read_bits(). Its strange as there are different functions for multiple registers read_register() and read_registers(). Someone, please address this issue.

ssshah86, 2018-08-23:
Modbus is asynchronous. it does not allow simultaneous communication over the network. This library does not support Modbus TCP. <- just getting this out of the way.

To answer your first question, if you would like to write to multiple coils then you can repeatedly call the `write_bit` function. See: [here](https://github.com/pyhys/minimalmodbus/blob/master/minimalmodbus.py#L212). It limits any input greater than 1. 

If you'd like to use Function Code 15, please provide a snippet of your code and error and I will try my best to help you. Make sure you're in debug mode. I've never used FC15. I think you can write up to 10 coils. FC15 [looks ](https://github.com/pyhys/minimalmodbus/blob/master/minimalmodbus.py#L668 ) supported. The problem may be writing `1`s with the _createBitPattern() method. See the comment in code [here](https://github.com/pyhys/minimalmodbus/blob/master/minimalmodbus.py#L1802). Try writing 0 first.

Devangsavla, 2018-08-31:
@ssshah86 Along with write, I also couldn't find multiple coil read. I might be missing out on something simple but please help me find out how to read multiple coils. Thanks. Minimalmodbus has saved a lot of my effort and the documentation is also rich.

ssshah86, 2018-08-31:
Can you write to device using `function code = 15` and post your results here? I can't do anything without that information. I don't have a slave device that supports FC15.

Following the code I think we start with the 2nd link in the post above where I state, FC 15 looks supported, [here](https://github.com/pyhys/minimalmodbus/blob/master/minimalmodbus.py#L668).

The payload looks to be generated. Try debug mode and post the contents of `payloadToSlave` here.

The method, `_numToTwoByteString` is never called in the #communicate section

alpintrekker, 2018-09-03:
@ssshah86, please correct me, if I'm wrong, but after browsing through the source code, it looks like FC15 supported only with single bit payload, which essentially the same as FC05.

This is a major weakness of this library, as there are Modbus slaves out on the playground, which requires _simultaneous_ coil writes to operate properly.

alpintrekker, 2018-09-03:
Well, I just learned about the internal function `[minimalmodbus._performCommand()](https://minimalmodbus.readthedocs.io/en/master/develop.html#extending-minimalmodbus)`. While it's intended for developer use and you should take care of payload generation, it allows any kind of play you can imagine. Enjoy!

ssshah86, 2018-09-04:
Yea I believe you are correct. I've been asking everyone who's trying to get FC15 implemented to run it in debug mode and provide the payload to me so I can do the math and fix the function , but no one has provided the data yet. I don't have a device that can take FC15. 

There's a comment [here](https://github.com/pyhys/minimalmodbus/blob/master/minimalmodbus.py#L1802) that asks if the payload that is being generated is correct.

ssshah86, 2018-09-04:
In the name of simplicity it may just be easier to use `pymodbus` since it seems FC 15 is supported, see here: https://pymodbus.readthedocs.io/en/latest/source/library/pymodbus.html#pymodbus.bit_write_message.WriteMultipleCoilsRequest

Devangsavla, 2018-09-04:
@ssshah86 I am asking help for reading multiple coil status. Not writing. FC 15 is for writing multiple coils not reading right?
Please help.

ssshah86, 2018-09-04:
Oh you're right, is that not FC02?

ddlyRabbit, 2019-05-07:
@ssshah86 FC02 can be used to read multiple coils but the payload preparation for FC02 uses NUMBER_OF_BITS which is set to one, does not matter what I pass into numberOfRegisters.

`        if functioncode in [1, 2]:
            payloadToSlave = _numToTwoByteString(registeraddress) + \
                            _numToTwoByteString(NUMBER_OF_BITS)`

I have not tried using this function to read multiple coils but seems like it won't work.

pyhys, 2019-07-28:
Yes, there is an obvious need for reading and writing mulitple coils (bits).

pyhys, 2019-08-06:
This will be included in next release

closed 2019-08-06 by pyhys

JanAlexanderPersonal, 2021-06-22:
Hi, is this problem solved somehow now?