pyhys/minimalmodbus #29: Control signals

adkolimp, 2018-05-14:
Hi,
In minimalmodbus I want to understand how the control lines are handled for MODBUS-RTU. In the example applications I couldn't see anything relevant to this. Any help is appreciated. 
I want to use minimalmodbus on Linkit7688 with OpenWRT. In my system I have connected the Rx and TX to UART1. I have connected RE and DI control pins to GPIO 4,5. My slave device is running Arduino UNO. I have connected the two with MAXIM RS485 (half duplex)connector.

ssshah86, 2018-08-23:
I'm not sure I understand your question. It sounds like you are focused on the lower level communications. You may be interested in the `_communicate` function. See [here](https://github.com/pyhys/minimalmodbus/blob/master/minimalmodbus.py#L802). The comments provide a good amount of information.

Also take a look at the [driver structure](https://minimalmodbus.readthedocs.io/en/master/develop.html#general-driver-structure)

adkolimp, 2018-08-31:
@ssshah86,
Thank you for pointing out to the documentation. We have moved onto some other topic right now. I am not too sure how to close this this issue. Sorry bothering you on this.

closed 2018-08-31 by adkolimp