pyhys/minimalmodbus #30: Problem to read rs485 using function 4 - Modbus Enron 32 bits

jhonatams, 2018-05-31:
Hi,

I'm using the below code to try read data from devide by rs485, but return checksum error.

The device use Modbus Enron 32 bits according of the documentation.
Device: [http://www.renzbr.com/produtos3.php?cod_produto=104](url)

> import minimalmodbus
import serial
instrument = minimalmodbus.Instrument('/dev/ttyUSB1', 1)
instrument.serial.baudrate = 19200  # Pode ser 9600 ou 19200.
instrument.serial.bytesize = 8  # Data bits.
instrument.serial.stopbits = 2  # Stop bits.
instrument.serial.parity = serial.PARITY_NONE
instrument.serial.timeout = 0.5
instrument.serial.dsrdtr = False
instrument.serial.rtscts = False
instrument.address = 1  # Pode ser de 1 até 247
instrument.debug = True
#instrument.precalculate_read_size = False
instrument.handle_local_echo = True
instrument.mode = minimalmodbus.MODE_RTU
instrument.read_register(registeraddress=6, functioncode=4)

But i receved the follow error:

>MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x01\x04\x00\x06\x00\x01\xd1\xcb' (01 04 00 06 00 01 D1 CB)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 1527799597297.7 ms, minimum silent period: 2.01 ms.
MinimalModbus debug mode. Discarding this local echo: '\x01\x04\x00\x06\x00\x01\xd1\xcb' (8 bytes).
MinimalModbus debug mode. Response from instrument: '\x01\x04\x04?G\xdb\xc1' (01 04 04 3F 47 DB C1) (7 bytes), roundtrip time: 101.6 ms. Timeout setting: 500.0 ms.

> Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/minimalmodbus.py", line 258, in read_register
    return self._genericCommand(functioncode, registeraddress, numberOfDecimals=numberOfDecimals, signed=signed)
  File "/usr/local/lib/python2.7/dist-packages/minimalmodbus.py", line 697, in _genericCommand
    payloadFromSlave = self._performCommand(functioncode, payloadToSlave)
  File "/usr/local/lib/python2.7/dist-packages/minimalmodbus.py", line 798, in _performCommand
    payloadFromSlave = _extractPayload(response, self.address, self.mode, functioncode)
  File "/usr/local/lib/python2.7/dist-packages/minimalmodbus.py", line 1075, in _extractPayload
    raise ValueError(text)
ValueError: Checksum error in rtu mode: '\xdb\xc1' instead of '\x08\xf3' . The response is: '\x01\x04\x04?G\xdb\xc1' (plain response: '\x01\x04\x04?G\xdb\xc1')

Searching about Enron Modbus I discovery that it's use 32bits by register instead 16 bits (not 2 registers 16bits to complete one 32bits).
I tried use read_long, read_float but nothing working too.

Anybody can help me?

I'm using Raspbian GNU/Linux 9 (stretch) operation system.

mtetreault, 2018-08-10:
You should be able to get it working by changing the _NUMBER_OF_BYTES_PER_REGISTER. https://github.com/pyhys/minimalmodbus/blob/master/minimalmodbus.py#L48

You will also need to make sure the communication function reads all the byte. You may need to modify the code to read the answer byte per byte, or at least in two shots. The first pass would get you the modbus header. The third byte of the header will give you the number of bytes left in the answer excluding the CRC.

You will probably also need to modify the decoding function to handle more bytes.

I hope this helps

pyhys, 2019-07-28:
Minimalmodbus unfortunately only support standard Modbus, which assumes 16 bits per register.

closed 2019-07-28 by pyhys