pyhys/minimalmodbus #32: ModBus on Linux Aria Board fails

Domi1993, 2018-08-20:
Hello

When i want to communicate with a device, but i gets only this Errors back (one Byte 0). I checked the Settings they are the same as on the device. what is the problem here?

It is a testing build (Bus is only 10cm long) and the device should answer me because it's was never in use before.
I use a "aria" [https://www.acmesystems.it/aria](url) with debain Jessie 8.9 installed

[image]
[image]

ssshah86, 2018-08-23:
Try `stopbits = 2`?

ssshah86, 2018-09-11:
I think the RAM is fine, it's running. What device are you communicating with?

ssshah86, 2018-09-11:
also try `gmaname = instrument.read_registers(0,34,3)`

ssshah86, 2018-09-11:
Try `gmaname = instrument.read_registers(40000, 34, 4)`. 

These are read only input registers?

pyhys, 2019-07-27:
This might be some hardware problem (termination etc). Maybe it is easier to find an answer on Stack Overflow from someone that knows about your board and instrument.

closed 2019-07-27 by pyhys