pyhys/minimalmodbus #36: write_register hangs when no slave is connected

austinstover, 2018-09-21:
### System information
- **OS Platform**: Windows 10
- **python version**: 3.6.5
- **minimalmodbus version**: 0.7.0
- **Directions to reproduce**: Use [com0com](http://com0com.sourceforge.net/) to create a virtual serial port (for testing purposes). Then run this code: 
```python
import minimalmodbus as mb

def main():
    mb.BAUDRATE = 9600
    mb.TIMEOUT = 0.1
    slave = mb.Instrument('COM8', 0)
    print("Start Write")
    slave.write_register(1, 0b100, functioncode=6) #Program hangs here
    print("Done")

if __name__ == '__main__':
    main()
```

### Problem
Minimalmodbus hangs when `write_register` is called without a slave connected. I tested this with a virtual serial port, but I'm pretty sure it also happened once when I was using an actual serial port and the Arduino microcontroller I was talking to was off.

pyhys, 2019-07-29:
The write() in pyserial is blocking by default. It is possible to define a write timeout, after which an exception is raised.

`instrument.serial.write_timeout=2.0`

See https://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.Serial.write

pyhys, 2019-08-03:
There will be an write timeout in next release.

closed 2019-08-03 by pyhys