pyhys/minimalmodbus #38: write_long() using function code 6

veeraragav, 2019-03-06:
1)The current implementation uses function code 16 to write two registers. My slave accepts only function code 6 to write. I need to write 32 bit value to two consecutive registers using your write_long() function using code 6. Can you add that feature?

2) When using read_long(address), the return value is calculated by placing value at address at upper half and value at address+1 at lower half. I need that to work vice-versa.

pyhys, 2019-07-27:
The byteorder described as your bullet 2) is a duplicate of #31

pyhys, 2019-07-28:
Function code 6 is for writing to a single register, so you need to yourself make two writes using that function code. It is unfortunate that your instrument does not support function code 16 when writing to more than one register is necessary for using it.

closed 2019-07-28 by pyhys