pyhys/minimalmodbus #41: Read bit seems to read two bits

haiyuehan, 2019-06-21:
I've got an instrument with bits at address 0 and address 1. 

When both those registers are 0, then everything is fine. However, if I write a 1 to both address 0 and 1, then I get an error when I try to read either one of the bits. 

The error happens in the _bitResponseToValue function; the parsed slave response would read 0x03, which is invalid. If I set address 1 to 0, but address 0 to 1, then I read 0x02. 

It seems like the library is reading both address 0 and 1 and reporting back. This error happens when I call instrument.read_bit on a AcuDC 240 meter.

pyhys, 2019-07-27:
When we ask the instrument for a single bit, it should maximally set a single bit in the response payload.  So it seem that the instrument not is following the standard.

I might come up with a workaround, but need some help with testing.

pyhys, 2019-07-28:
This will be resolved when #27 is implemented, because then you can read the two addresses simutaneously. It seems to be a problem in your instrument when reading a single bit.

closed 2019-07-28 by pyhys