pyhys/minimalmodbus #45: Issue in subsequent read of 2 registers

amitrana3348, 2020-02-11:
When I read a register for e.g. 201, due to physical wire error, am doing it in for loop for 4 times. 

Now when I read next register, e.g. 209; again using a for loop for 4 times. What happens is the readings for register 201 are read while 209 also, for the first time only, when subsequent for loops execute, I get the new reading.
But for first time, it definitely reads the last value only. I tried with numerous trials but it appears the same way.

below is a snippet
`for x in range(0,4):
          try:
               print('reading Solar Kvarh')
               pfc3 = meter.read_registers(209,6,3)
               pf1 = pfc3[0]
               pf2 = pfc3[1]
               solar_kvarh = unpack('<f',pack('<HH',pf1,pf2))[0]
               solar_kvarh = round(solar_kvarh,4)
               print 'Solar kvarh = ',solar_kvarh
          except IOError:
               print("Failed to read from instrument")
          except ValueError:
               print("value error")
          time.sleep(2)
     print("*************************************************************************************\n")
     
     for x in range(0,4):
          try:
               print('reading Solar Kvah')
               pfc4 = meter.read_registers(217,6,3)
               pf1 = pfc4[0]
               pf2 = pfc4[1]
               solar_kvah = unpack('<f',pack('<HH',pf1,pf2))[0]
               solar_kvah = round(solar_kvah,4)
               print 'Solar kvah = ',solar_kvah
          except IOError:
               print("Failed to read from instrument")
          except ValueError:
               print("value error")
          time.sleep(2)
`
first time, the KVAH reads the same value as that of KVARH in above code
from next read, it comes back to reading the expected reading

xavipolo, 2020-04-21:
May be your device needs 
`instrument.close_port_after_each_call = True`

pyhys, 2021-08-08:
Formatted the code in the question.
Closing due to inactivity.

closed 2021-08-08 by pyhys