pyhys/minimalmodbus #46: 'minimalmodbus' has no attribute 'instrument'

joshuacox, 2020-03-14:
Ran into this firing up interactive mode:

```
$ python                                                                                            
Python 3.8.1 (default, Jan 22 2020, 06:38:00) 
[GCC 9.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import minimalmodbus
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/thoth/monitaur/daijin-modbus/minimalmodbus.py", line 4, in <module>
    instrument = minimalmodbus.Instrument('/dev/ttyUSB0', 1)  # port name, slave address (in decimal)
AttributeError: partially initialized module 'minimalmodbus' has no attribute 'Instrument' (most likely due to a circular import)
```

possibly my python is too new?

```
python -V                                                                                                              [8:12:41]
Python 3.8.1
```

xavipolo, 2020-04-21:
I'm working with python 3.7.4 without problems.

j123b567, 2020-05-05:
Do you have your custom file with name `minimalmodbus.py` and you try to import system library `minimalmodbus`? This can cause lot of problems. Just rename your file.

pyhys, 2021-08-08:
Thanks @j123b567 
Closing due to inactivity

closed 2021-08-08 by pyhys