pyhys/minimalmodbus #51: Changing byte order in CRC checksum

maxelot, 2020-04-27:
I am currently working with a device that is using modbus as a base for its own protocol with own function codes. Also, they changed the order of bytes in the CRC checksum. 
I saw that the function 
`def _num_to_twobyte_string(value, number_of_decimals=0, lsb_first=False, signed=False)`
has the option to switch the bytes. From my understanding the function
`def _calculate_crc_string(inputstring):`
is used to create and check all CRC checksums. Here it is hardcoded that the LSB is send first.
`return _num_to_twobyte_string(register, lsb_first=True)`

Is there an option how I can quickly jump between "CRC-LSB first" and "CRC-HSB first" when using devices that use both the switched and the normal CRC checksum on the same bus?

j123b567, 2020-05-05:
You can always do something like this. 

```python
import minimalmodbus

def _calculate_crc_string(inputstring):
    """Calculate CRC-16 for Modbus.

    Args:
        inputstring (str): An arbitrary-length message (without the CRC).

    Returns:
        A two-byte CRC string, where the least significant byte is first.

    """
    minimalmodbus._check_string(inputstring, description="input CRC string")

    # Preload a 16-bit register with ones
    register = 0xFFFF

    for char in inputstring:
        register = (register >> 8) ^ minimalmodbus._CRC16TABLE[(register ^ ord(char)) & 0xFF]

    return minimalmodbus._num_to_twobyte_string(register, lsb_first=False)

minimalmodbus._calculate_crc_string = _calculate_crc_string
```

But it is funny how the CRC is misunderstood. The nice property of correctly used CRC16 is that 

`crc16(message) == 0` and
`crc16(message[:-2]) == two_byte_to_uint16(message[-2:])`
assuming the `message` has last two bytes its own crc16 like in modbus-rtu
If you swap (as the developer of some protocol) the endianity of the CRC, you loos the first case.

pyhys, 2021-08-08:
Thanks @j123b567 for the response.

closed 2021-08-08 by pyhys