pyhys/minimalmodbus #52: Using special function codes

luixal, 2020-04-30:
Hi,

**DISCLAIMER**: I'm quite new to RS485 so... maybe I'm asking something really stupid :)

I've been working on a _"driver"_ for reading values and controlling an Ozone Sensor that uses Modbus over a RS485 port.

Everything has been quite smooth by using this (great, by the way) library. Main features of the sensor use the standar function codes (1, 3, 4, 6, 16...) but it also uses function code 15 to configure a couple of relays (INT16 for 0 or 1 values). Here is the reference from the specs (columns are address, description, function code, R/W, length, value type, possible values and default values):

[image]

Is it possible to use those function code in any way? Does it make sense?

Thanks!

j123b567, 2020-05-05:
Function code 15 is standard command "Write Multiple Coils" implemented by `write_bits`. So it seems nothing special in this case.

luixal, 2020-05-07:
Ok, I guess I was trying to use another function forcing function code 15 to read data from those registeres. I'll try with `read_bits` and `write_bits`.

Thanks!

closed 2021-08-08 by pyhys