pyhys/minimalmodbus #54: cannot change serial.parity

rma-ln, 2020-06-09:
The following code works correctly:
`import minimalmodbus   # pip3 install -U minimalmodbus`
`device = minimalmodbus.Instrument('/dev/ttyS0', 7)`
`device.serial.baudrate = 9600`
`device.serial.bytesize = 8`
`device.serial.parity = 'N'`
`device.serial.stopbits = 1`
`print(device)`

An attempt to change the serial.parity 'N' to e.g. 'E' or 'O' ends in an error:
`import minimalmodbus   # pip3 install -U minimalmodbus`
`device = minimalmodbus.Instrument('/dev/ttyS0', 7)`
`device.serial.baudrate = 9600`
`device.serial.bytesize = 8`
`device.serial.parity = 'E'`  # <-- place of change, possible values 'N', 'E', 'O', 'M', 'S'
`device.serial.stopbits = 1`

[iflag, oflag, cflag, lflag, ispeed, ospeed, cc])
termios.error: (22, 'Invalid argument')

[image]

it's probably a minimalmodbus module error

j123b567, 2020-06-09:
This is definitely not a problem of minimalmodbus. Minimalmodbus is using [pyserial](https://github.com/pyserial/pyserial) module, which is responsible for this configuration and this error message.

If you have similar problems, please always report operating system, python version and hardware you are using.
Because it seems to be related to some crappy hardware which lacks support for setting parity, e.g. [mini UART on RaspberryPi](https://www.raspberrypi.org/documentation/configuration/uart.md), so just use/configure other serial port or use USB/serial converter.

If it is not your case, please report this to pyserial project.

rma-ln, 2020-06-09:
You're right. When you change the serial port to Moxa NPort, everything works fine. Thank you for your help and very quick response.

closed 2020-06-09 by rma-ln