pyhys/minimalmodbus #55: Communication work with v0.7 but not with v1.0.2

viewpointsa, 2020-07-07:
This is my code:
```
minimalmodbus.BAUDRATE = 9600
minimalmodbus.PARITY = serial.PARITY_EVEN
minimalmodbus.BYTESIZE = 8
minimalmodbus.STOPBITS = 1
minimalmodbus.TIMEOUT = 1
self.instrument = minimalmodbus.Instrument(port,slaveaddress,mode='rtu')        
self.instrument.debug = True
self.instrument.read_float(address)
```
Output from version 0.7:

```
MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x01\x03\x00\x04\x00\x01\xc5\xcb' (01 03 00 04 00 01 C5 CB)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 1594126730633.0 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '\x01\x03\x02\x00\x00\xb8D' (01 03 02 00 00 B8 44) (7 bytes), roundtrip time: 29.0 ms. Timeout setting: 1000.0 ms.
MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x01\x03\x00\x00\x00\x01\x84\n' (01 03 00 00 00 01 84 0A)
MinimalModbus debug mode. Sleeping for 3.0 ms. Minimum silent period: 4.0 ms, time since read: 1.0 ms.
MinimalModbus debug mode. Response from instrument: '\x01\x03\x02\x00\xdb\xf8\x1f' (01 03 02 00 DB F8 1F) (7 bytes), roundtrip time: 30.0 ms. Timeout setting: 1000.0 ms.
Temperature 21.9 Centigrade
```
And with version 1.0.2:
```
MinimalModbus debug mode. Will write to instrument (expecting 7 bytes back): '\x01\x03\x00\x04\x00\x01\xc5\xcb' (01 03 00 04 00 01 C5 CB)
MinimalModbus debug mode. Clearing serial buffers for port COM4
MinimalModbus debug mode. No sleep required before write. Time since previous read: 1594126681759.00 ms, minimum silent period: 2.01 ms.
MinimalModbus debug mode. Response from instrument: '' () (0 bytes), roundtrip time: 60.0 ms. Timeout for reading: 50.0 ms.
```
Test with python 3.6 and 2.7 on Windows 10 x64 with USB-RS485-WE-1800-BT. Buffer are the same (01 03 00 04 00 01 C5 CB)

ddx000, 2020-08-28:
Same issue here, 

minimalmodbus.BAUDRATE = 9600
minimalmodbus.PARITY = 'N'
minimalmodbus.BYTESIZE = 8
minimalmodbus.STOPBITS = 1
minimalmodbus.TIMEOUT = 0.5

instrument = minimalmodbus.Instrument('COM3',1)
ModList = instrument.read_registers(0, 10, functioncode=4)

v0.7 worked, but not with v1.0.2

pyhys, 2021-08-08:
There is no response from the instrument, according to the printout. This has probably to do with some settings for the serial port in Windows. It would great if someone could try to reproduce this, and experiment with the serial port settings.

hamdanal, 2021-12-25:
The source of the issue is that in version 0.7, you can set module level constants (such as `BAUDRATE` and `PARITY`) and these constants will be used as defaults for the `serial.Serial` object created; see: https://github.com/pyhys/minimalmodbus/blob/e99f4d74c83258c6039073082955ac9bed3f2155/minimalmodbus.py#L103
In version 1.0.0+, this is no longer true as the defaults are hardcoded in the `Serial` initializer; see: https://github.com/pyhys/minimalmodbus/blob/3e04bfff450d777ef5b137036e1ccff20535b32b/minimalmodbus.py#L208-L216
This means the settings you set using `minimalmodbus.<SERIAL_CONSTANTS>` will have no real effect on the configuration of the `minimalmodbus.Instrument` instance created. To solve the issue change your code to:
```python
self.instrument = minimalmodbus.Instrument(port,slaveaddress,mode='rtu')
self.instrument.baudrate = 9600
self.instrument.parity = serial.PARITY_EVEN
self.instrument.bytesize = 8
self.instrument.stopbits = 1
self.instrument.timeout = 1
self.instrument.debug = True
self.instrument.read_float(address)
```
This works in all versions.

pyhys, 2022-02-17:
You are right @hamdanal 
Thanks for the help.

Closing.

closed 2022-02-17 by pyhys