pyhys/minimalmodbus #58: Issue with large register addresses

tr-gis, 2020-09-04:
I am new to Modbus communication. I am using minimalmodbus library to read/write device registers from a delta plc. Unsigned int is currently used in the library for register addresses, allowing one to specify values from 0 to 65535. However, the registers that I want to access have addresses much larger than this. Is there a workaround, like increasing size from unsigned int to unsigned long int?

[image]

The yellow highlighted part is the device register that I am interested in. You can see that addresses range from 400001 to 430000. I am attaching function code mapping below for further reference.

[image]

arthurtarasov, 2020-09-04:
Those look like holding registers 1-30000. Try dropping 4 and talking to an address within 1-30000 and see if it responds

tr-gis, 2020-09-04:
That works!. Thank you. Is it safe to assume that I can follow the same addressing way for other registers as well ? ie; drop the leftmost digit. I was not sure if they are holding registers since generally holding registers have address space 40001 - 49999, atlteast that is what wikipedia says.

[image]

arthurtarasov, 2020-09-04:
It seems so. The first digit is 0-4 and what follows is no larger than a 2-byte address in every case. I'm not sure if this is the conventional way so other applications might get confused by that. But if you test it and it works than it should be safe enough for that case.

closed 2020-09-05 by tr-gis