pyhys/minimalmodbus #59: Simple modbus write to register example

krupis, 2020-10-29:
Hey. I have connected RX/TX pins of raspberry PI to signal analyzer and testing the sginal with the code:
```
# \x02 - trigger scanning
# \x03 after scanning is done
import serial

sendSerial = serial.Serial ("/dev/serial0", 9600)

sendSerial.write (str.encode("\x03"))
sendSerial.close()
```
the message goes through and I am able to capture it

I am now trying to implement simmilar thing using this MODBUS library and the code:
```

#!/usr/bin/env python3
import minimalmodbus

instrument = minimalmodbus.Instrument('/dev/serial0', 1)  # port name, slave address (in decimal)

## Change temperature setpoint (SP) ##
NEW_TEMPERATURE = 95
instrument.write_register(10, NEW_TEMPERATURE, 1)  # Registernumber, value, number of decimals 
```
```
Traceback (most recent call last):
  File "/home/pi/Desktop/lukas_programming/PTL_python/minimalmodbus_Test.py", line 12, in <module>
    instrument.write_register(10, NEW_TEMPERATURE, 1)  # Registernumber, value, number of decimals
  File "/home/pi/.local/lib/python3.7/site-packages/minimalmodbus.py", line 518, in write_register
    payloadformat=_PAYLOADFORMAT_REGISTER,
  File "/home/pi/.local/lib/python3.7/site-packages/minimalmodbus.py", line 1170, in _generic_command
    payload_from_slave = self._perform_command(functioncode, payload_to_slave)
  File "/home/pi/.local/lib/python3.7/site-packages/minimalmodbus.py", line 1240, in _perform_command
    response = self._communicate(request, number_of_bytes_to_read)
  File "/home/pi/.local/lib/python3.7/site-packages/minimalmodbus.py", line 1406, in _communicate
    raise NoResponseError("No communication with the instrument (no answer)")
minimalmodbus.NoResponseError: No communication with the instrument (no answer)
```

WHat could be the issue here? I am just trying to probe the TX/RX signals and see what kind of message does it send.

Also, how are direction control pin is being controlled by this library? I have also connected one of the GPIO from the raspberry to the DIR pin of my RS485 driver . But first I just want to make sure I am able to capture the signal that I am trying to send

ccall48, 2021-02-11:
You also need to pass it the write function code, most of the devices i have played with it is 6.
https://minimalmodbus.readthedocs.io/en/stable/apiminimalmodbus.html#minimalmodbus.Instrument.write_register

pyhys, 2021-08-08:
Modbus usage questions for specific applications might find better responses on Stack Overflow. 
See https://minimalmodbus.readthedocs.io/en/stable/troubleshooting.html#support for some tips

closed 2021-08-08 by pyhys