pyhys/minimalmodbus #62: Checksum error

GerfriedC, 2021-01-17:
I changed the number of bits to 16 in the controller but still I get:
`Checksum error in rtu mode: '\x01ú' instead of 'éú' .`
The response is: '\x01\x03\x9cA\x00\x01ú' (plain response: '\x01\x03\x9cA\x00\x01ú')
many thanks in advance

j123b567, 2021-01-17:
What bits in which controller? 

Minimalmodbus do it right, so your instrument is broken. CRC should be 0xFA 0xE9 which is exactly what minimalmodbus complains.

pyhys, 2021-08-08:
Closing due to inactivity.

Thanks @j123b567 for the support.

closed 2021-08-08 by pyhys

themzlab, 2025-01-29:
people may find this resource handy.  it can parse a modbus packet and confirm the CRC
https://rapidscada.net/modbus/

also, there are some pain points in modbus.  the word order is not standardized.  the CRC is 2 bytes and they are appended in this order.  
.... low byte of CRC, high byte of CRC.  

This might be the opposite order of your payload.  Furthermore when requesting registers you do so in 16-bit words.   There are other contexts where the data is counted as bytes - so this is 2X the word count.

these are just things to check as they have tripped me up on this same issue.