pyhys/minimalmodbus #63: How to measure round trip time

tr-gis, 2021-02-11:
Is there a function among the APIs that will return round trip time for a function call?.

ccall48, 2021-02-11:
https://minimalmodbus.readthedocs.io/en/stable/apiminimalmodbus.html#minimalmodbus.Instrument.debug

From memory this gives transaction time.

tr-gis, 2021-02-12:
Hi @ccall48 . I have seen roundtrip times displayed as part of debug output. I am looking for function calls that will return round trip time, with debug option set to True, it only prints the time.

pyhys, 2021-08-08:
@tr-gis There is unfortunately no plans to adjust the official API with this functionality, but you can yourself write to some new attribute of the `Instrument` each time you get a response.

closed 2021-08-08 by pyhys

pyhys, 2021-08-10:
I did actually include this in the latest release:
https://minimalmodbus.readthedocs.io/en/stable/apiminimalmodbus.html#minimalmodbus.Instrument.roundtrip_time