pyhys/minimalmodbus #65: help needed read register low high byte

riker65, 2021-03-22:
Hi 
I am having a register to read but can not manage to format is correct

any hints?

Thanks T

`Adr: 0103H
Byte: 2
Read:
describtion: The internal 
temperature of the 
controller (high 8.
bits)
The actual temperature 
value (internal two
-way 
temperature, transmission 
of one high).
(b7: symbol bit; b0
-b6
temperature value).
℃`

j123b567, 2021-03-22:
First question: It is outside of the scope of this library. Without propper documentation, some context and example values, nobody can help you.

Second question: `read_string` does not work for you?
```python
instrument.read_string(0xc, 8)
```

riker65, 2021-03-22:
> Second question: `read_string` does not work for you?
> 
> ```python
> instrument.read_string(0xc, 8)
> ```

Thanks, great could read the string with: 
`model = instrument.read_string(0xc,8)`
**-> '  RBC50D1S-G1   '**

even  documentation is saying  
`read_string(registeraddress, number_of_registers=16, functioncode=3)` 
so i do not understand your "short" version.

also worked 
`instrument.read_string(12,8, 3)`

so I assume function 3 is default for read.....

work on the example for the first question.

will update

j123b567, 2021-03-22:
> even documentation is saying
> `read_string(registeraddress, number_of_registers=16, functioncode=3)`
> so i do not understand your "short" version.

Please, learn a little bit of python and [default argument values](https://docs.python.org/3/tutorial/controlflow.html#default-argument-values)

riker65, 2021-03-22:
concerning the 2. topic:

try to clarify more the topic>:

Address: 104H Bytes: 2  Read  
Description: controller inside temperature(high 8 bit) and 
auxilliary battery temperature（low 8 bit）
b7:sign bit；b0-b6 temperature)

Temperature is a register which I could not read/format correct as it has high and low bytes with a padded bit for temp polarity.

unfortunaltely my bit and byte binary knowledge is limited.

when I try to read the 2 different temperatures in the 2 byte 
```
controllerTemp  =  instrument.read_register(259)
auxTemp  =  instrument.read_register(259)
```

I get  for example this:

('controllerTemp :', 4363)
('auxkTemp   :', 4363)

do not know how to separate the high and low and the b7:sign bit；b0-b6 temperature)

j123b567, 2021-03-22:
It is really out of the scope of the discussion here to teach you programming in python.
Learn about bitwise-operators in python and you will then clearly see, that extraction of the temperature is just sequence of bitwise right shift	and bitwise AND operations.

riker65, 2021-03-23:
not an issue of the library but knowledge of myself

closed 2021-03-23 by riker65