pyhys/minimalmodbus #67: Long wait time between receiving a reply and sending the next request

JoeEarlam, 2021-05-14:
I have a short script that requests some registers from two clients on the same serial adapter. Functionally I have data transferring with no issues, but looking at the bus on an oscilloscope the bus utilisation is very low. Requests are only being sent every 17ms, despite my client responding to the request within 5ms. This leaves ~10ms of dead time between receiving a packet and sending the next request, regardless of baud rate (assuming it's fast enough to do all the comms within that 17ms period).

Is this something that is configurable?

Thanks!

`#!/usr/bin/env python3
import minimalmodbus
import serial

instrument1 = minimalmodbus.Instrument('/dev/ttyUSB0', 1)  # port name, client address (in decimal)
instrument1.serial.baudrate = 115200         # Baud
instrument1.serial.bytesize = 8
instrument1.serial.parity   = serial.PARITY_EVEN
instrument1.serial.stopbits = 1
instrument1.serial.timeout  = 0.05          # seconds

instrument2 = minimalmodbus.Instrument('/dev/ttyUSB0', 2)  # port name, client address (in decimal)
instrument2.serial.baudrate = 115200         # Baud
instrument2.serial.bytesize = 8
instrument2.serial.parity   = serial.PARITY_EVEN
instrument2.serial.stopbits = 1
instrument2.serial.timeout  = 0.05          # seconds

## Read values
while(1):
  print("Client 1:", end=' ')
  value = instrument1.read_registers(0,16)  # Registernumber, number of registers
  print(value)

  print("Client 2:", end=' ')
  value = instrument2.read_registers(0,16)  # Registernumber, number of registers
  print(value)

`

pyhys, 2021-08-10:
Hi!
In the latest version of MinimalModbus (2.0.0) there is functionality to measure the roundtrip time. See https://minimalmodbus.readthedocs.io/en/stable/apiminimalmodbus.html#minimalmodbus.Instrument.roundtrip_time

JoeEarlam, 2021-08-12:
Thanks, that's a useful feature but it's not what I'm talking about - the round trip is quick as the MCU on the other end of the bus responds quickly, the issue is the wait time between the PC receiving data and sending the next packet.

Is this a Python or OS limitation? I am not particularly familiar with high level languages or programming for OSes.

laran002, 2021-12-01:
I am having the same issue. In my case the delay is in the range of 10-25ms. I have not been able to isolate the problem. However, continuous writing directly to the serial port does not behave like this.

j123b567, 2021-12-01:
Can you please specify
 - python version
 - OS - linux distro and version, windows version, ...
 - hardware - PC, SBC (architecture/chip)
 - pyserial version
 - serial port HW - e.g. native, USB (TFDI, CH340, CP2102, ...), ...

JoeEarlam, 2021-12-01:
Python version: 3.8.10
OS: Ubuntu 20.04 X64_86
Hardware: PC
pyserial version: 3.5
serial port hardware: FTDI 3.3V RS232 converter, ID 0403:6001 with Seeed RS485 converter (MAX13487).

laran002, 2021-12-01:
**Python version:** 3.7.3
**OS:** Raspbian GNU/Linux 10 (buster) 
**Hardware:** RPi 4B
**PySerial:** version 3.5
**Serial port Hardware:**  ISO3080 (Isolated 5-V Full- and Half-Duplex RS-485 Transceiver) on i/o pins I have also redirected the HW uart from bluetooth to pin 14 & 15 to ensure the default SW uart is not causing this. 

### Please find my initialization code below:
#### Create device
motor = minimalmodbus.Instrument(  '/dev/serial0', 1)  # port name, slave address (in decimal)
motor.mode = minimalmodbus.MODE_RTU   # rtu or ascii mode
motor.clear_buffers_before_each_transaction = True # Is this required?

#### Setup Modbus communication properties
motor.serial.close()
motor.serial.baudrate = 57600
motor.serial.bytesize = 8
motor.serial.parity   = serial.PARITY_NONE
motor.serial.stopbits = 1
motor.serial.timeout = 0.5
motor.serial.open()

laran002, 2021-12-02:
Just for reference, I tested using PyModbus 2.5.3 and that did not show the same delay.
Also, I measured execution time of read_long() and write_long() and that corresponds to the delay´s I am noticing on the bus.