pyhys/minimalmodbus #71: Minimalmodbus don't write values above 1000

aldosola, 2021-07-23:
I'm trying to write values above 1000 using the minimalmodbus library, but my code generates the following error:

```
Traceback (most recent call last):
  File "D:/Projetos/ON/HMI/modbus_tk_test.py", line 19, in <module>
    instrument.write_register(38, 1001)
  File "D:\Projetos\ON\HMI\venv\lib\site-packages\minimalmodbus.py", line 511, in write_register
    self._generic_command(
  File "D:\Projetos\ON\HMI\venv\lib\site-packages\minimalmodbus.py", line 1177, in _generic_command
    payload_from_slave = self._perform_command(functioncode, payload_to_slave)
  File "D:\Projetos\ON\HMI\venv\lib\site-packages\minimalmodbus.py", line 1250, in _perform_command
    payload_from_slave = _extract_payload(
  File "D:\Projetos\ON\HMI\venv\lib\site-packages\minimalmodbus.py", line 1776, in _extract_payload
    _check_response_slaveerrorcode(response)
  File "D:\Projetos\ON\HMI\venv\lib\site-packages\minimalmodbus.py", line 3436, in _check_response_slaveerrorcode
    raise error
minimalmodbus.IllegalRequestError: Slave reported illegal data value
```
If I use values below 1001 no error is generated. Here is my code:

```
import minimalmodbus
import serial.serialutil
import time

instrument = minimalmodbus.Instrument('COM4', 247, minimalmodbus.MODE_RTU)

instrument.serial.baudrate = 115200
instrument.serial.bytesize = 8
instrument.serial.parity = serial.PARITY_NONE
instrument.serial.stopbits = 1
instrument.serial.timeout = 1          # seconds
instrument.clear_buffers_before_each_transaction = False
instrument.close_port_after_each_call = False
instrument.precalculate_read_size = False
print(minimalmodbus._calculate_minimum_silent_period(115200))

while True:
    data = instrument.read_registers(38, 1, 3)
    instrument.write_register(38, 1001)
    print(data)
    time.sleep(minimalmodbus._calculate_minimum_silent_period(115200))
```
Any help is appreciated.

j123b567, 2021-07-23:
As you can see `Slave reported illegal data value` so it is limitation of your slave device, not a problem of this library.

pyhys, 2021-08-08:
Thanks @j123b567

closed 2021-08-08 by pyhys