pyhys/minimalmodbus #72: Modbus TCP implementation

ihategoto, 2021-07-26:
Hello,
I'm currently writing a Modbus TCP client using Minimalmodbus utilities. 
The task turned out to be fairly easy as Minimalmodbus provides all the utilities to create the Modbus PDU;
all I had to do was writing utilities to create the MBAP header and the socket management.
My goal is to integrate the Modbus TCP implementation with the current one.
I've been testing the code that I wrote and it seems to behave properly. 
I'm writing here because I noticed that the Modbus TCP  implementation is not planned for future releases, and my intention was, as soon as the code is ready, to submit a pull request to merge my fork and the original one. Might that be a good idea?

PS: feel free to contribute to my code.

pyhys, 2021-08-08:
Hi, thanks for your interest in MInimalModbus.
It is planned than in the future modify the internal representation of the data, now that Python2 support is dropped. So instead of storing data as strings, the data will be stored as bytes. That is a major rewrite of the internals. 
Any TCP functionality needs to be done after that.

thatdecade, 2025-08-08:
I have a fork with Modbus TCP support in MinimalModbus. 

https://github.com/2G-Engineering/minimalmodbus/blob/ec615e01c47e7fa56e68da4eed555047a19c9c6d/minimalmodbus.py

* No breaking changes. RTU and ASCII paths are untouched.
* TCP is enabled by passing a connection string as the port: `Instrument(port="tcp://192.168.1.100:502", slaveaddress=1)`