pyhys/minimalmodbus #79: minimalmodbus Yocto recipe fails

lemmi25, 2021-11-26:
Hi,

I am trying to use minimalmodbus in yocto.

When I am following the recipe instructions on the [manual page](https://minimalmodbus.readthedocs.io/en/stable/advancedusage.html#including-minimalmodbus-in-a-yocto-build) I get this error after running 

`bitbake core-image-base `

```
ERROR: ParseError at /home/build/workspace/recipes/python3-minimalmodbus/python3-minimalmodbus_git.bb:18: Could not inherit file classes/distutils.bbclass    | ETA:  --:--:--

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

I also used the devtool to auto generate a recipe for me. It looks like this:

```
# Recipe created by recipetool
# This is the basis of a recipe and may need further editing in order to be fully functional.
# (Feel free to remove these comments when editing.)

# WARNING: the following LICENSE and LIC_FILES_CHKSUM values are best guesses - it is
# your responsibility to verify that the values are complete and correct.
#
# The following license files were not able to be identified and are
# represented as "Unknown" below, you will need to check them yourself:
#   LICENSE
LICENSE = "Unknown"
LIC_FILES_CHKSUM = "file://LICENSE;md5=27da4ba4e954f7f4ba8d1e08a2c756c4"

SRC_URI = "git://github.com/pyhys/minimalmodbus.git;protocol=https;branch=master"

# Modify these as desired
PV = "1.0+git${SRCPV}"
SRCREV = "185afffd2d65b585f69bc5e3f640270d2bd27444"

S = "${WORKDIR}/git"

# NOTE: this is a Makefile-only piece of software, so we cannot generate much of the
# recipe automatically - you will need to examine the Makefile yourself and ensure
# that the appropriate arguments are passed in.

do_configure () {
	# Specify any needed configure commands here
	:
}

do_compile () {
	# You will almost certainly need to add additional arguments here
	oe_runmake
}

do_install () {
	# NOTE: unable to determine what to put here - there is a Makefile but no
	# target named "install", so you will need to define this yourself
	:
}

```

But the parts in **do_configure**, **do_compile** and **do_install** are not filled out. Can maybe someone help me here or provide a working recipe. 

Many thanks

Moritz

j123b567, 2021-11-26:
Isn't it just incompatibility between python2 based and python3 based build?

I'm not using Yocto but just googling.

If you use the bb file from the tutorial, but replace

`distutils` -> `distutils3`
`python-pyserial` -> `python3-pyserial`

And also all versions and hashes. It should at least move somewhere.

lemmi25, 2021-11-26:
Thanks for the fast reply!

this is my recipe (with your changes):

```
SUMMARY = "Easy-to-use Modbus RTU and Modbus ASCII implementation for Python"
SECTION = "devel/python"
LICENSE = "CLOSED"
LIC_FILES_CHKSUM = ""

DEPENDS = "python3"
RDEPENDS_${PN} = "python3-pyserial"

PR = "r0"

SRC_URI = "${SOURCEFORGE_MIRROR}/project/minimalmodbus/${PV}/MinimalModbus-${PV}.tar.gz"

SRC_URI[md5sum] = "1b2ec44e9537e14dcb8a238ea3eda451"
SRC_URI[sha256sum] = "d9acf6457bc26d3c784caa5d7589303afe95e980ceff860ec2a4051038bc261e"

S = "${WORKDIR}/MinimalModbus-${PV}"

inherit distutils3
```

This is the error:

```
WARNING: python3-minimalmodbus-2.0.1-r0 do_fetch: Failed to fetch URL https://downloads.sourceforge.net/project/minimalmodbus/2.0.1/MinimalModbus-2.0.1.tar.gz, attempting MIRRORS if available
ERROR: python3-minimalmodbus-2.0.1-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/home/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/recipe-sysroot-native/usr/bin/python3-native:/opt/yocto/poky/scripts:/home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/recipe-sysroot/usr/bin/crossscripts:/home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/recipe-sysroot-native/usr/sbin:/home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/recipe-sysroot-native/usr/bin:/home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/recipe-sysroot-native/sbin:/home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/recipe-sysroot-native/bin:/opt/yocto/poky/bitbake/bin:/home/build/tmp/hosttools"; export HOME="/home/build"; /usr/bin/env wget -t 2 -T 30 --passive-ftp --no-check-certificate -P /home/build/downloads 'https://downloads.sourceforge.net/project/minimalmodbus/2.0.1/MinimalModbus-2.0.1.tar.gz' --progress=dot -v failed with exit code 8, no output
ERROR: python3-minimalmodbus-2.0.1-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'https://downloads.sourceforge.net/project/minimalmodbus/2.0.1/MinimalModbus-2.0.1.tar.gz')
ERROR: Logfile of failure stored in: /home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/temp/log.do_fetch.77360
ERROR: Task (/home/custom-layer/meta-custom-python/recipes-python/python/python3-minimalmodbus_2.0.1.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4791 tasks of which 4790 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/custom-layer/meta-custom-python/recipes-python/python/python3-minimalmodbus_2.0.1.bb:do_fetch
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

So I thought I should get get the soureces from somewere else, but still:

recipe:

```
SUMMARY = "Easy-to-use Modbus RTU and Modbus ASCII implementation for Python"
SECTION = "devel/python"
LICENSE = "CLOSED"
LIC_FILES_CHKSUM = ""

DEPENDS = "python3"
RDEPENDS_${PN} = "python3-pyserial"

PR = "r0"

SRC_URI = "https://files.pythonhosted.org/packages/78/99/8cd22b4465e697bae2b02fd06aaccd4c5cdfbb18945d728db99f23d71df9/minimalmodbus-2.0.1.tar.gz"

#SRC_URI[md5sum] = "1b2ec44e9537e14dcb8a238ea3eda451"
SRC_URI[sha256sum] = "cf873a2530be3f4b86467c3e4d47b5f69fd345d47451baca4adbf59e2ac36d00"

S = "${WORKDIR}/MinimalModbus-${PV}"

inherit distutils3
```

I am getting this error:

```
Sstate summary: Wanted 8 Found 0 Missed 8 Current 2593 (0% match, 99% complete)
NOTE: Executing Tasks
ERROR: python3-minimalmodbus-2.0.1-r0 do_compile: 'python3 setup.py build ' execution failed.
ERROR: python3-minimalmodbus-2.0.1-r0 do_compile: Execution of '/home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/temp/run.do_compile.77467' failed with exit code 1
ERROR: Logfile of failure stored in: /home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/temp/log.do_compile.77467
Log data follows:
| DEBUG: Executing shell function do_compile
| /home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/recipe-sysroot-native/usr/bin/python3-native/python3: can't open file 'setup.py': [Errno 2] No such file or directory
| ERROR: 'python3 setup.py build ' execution failed.
| WARNING: /home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/temp/run.do_compile.77467:1 exit 1 from 'exit 1'
| ERROR: Execution of '/home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/temp/run.do_compile.77467' failed with exit code 1
ERROR: Task (/home/custom-layer/meta-custom-python/recipes-python/python/python3-minimalmodbus_2.0.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4727 tasks of which 4726 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/custom-layer/meta-custom-python/recipes-python/python/python3-minimalmodbus_2.0.1.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

Which is not to wrong because there is no setup.py in minimalmodbus (At least in the root directory). Which brings me back to the first recipe I showed in this issue. In that you have to give instructions on how to build it based on a make file.

I hope this clarifies the problem a little bit more.

Many thanks again Moritz

j123b567, 2021-11-26:
so what about

python3-minimalmodbus_2.0.1.bb
```
SUMMARY = "Easy-to-use Modbus RTU and Modbus ASCII implementation for Python"
SECTION = "devel/python"
LICENSE = "Apache-2.0"
LIC_FILES_CHKSUM = "file://LICENCE;md5=27da4ba4e954f7f4ba8d1e08a2c756c4"

DEPENDS = "python3"
RDEPENDS_${PN} = "python3-pyserial"

SRCNAME = "minimalmodbus"
SRC_URI = "https://pypi.python.org/packages/source/m/${SRCNAME}/${SRCNAME}-${PV}.tar.gz"

SRC_URI[md5sum] = "3fe320f7be761b6a2c3373257c431c31"
SRC_URI[sha256sum] = "cf873a2530be3f4b86467c3e4d47b5f69fd345d47451baca4adbf59e2ac36d00"

S = "${WORKDIR}/${SRCNAME}-${PV}"

inherit distutils3
```

lemmi25, 2021-11-26:
Hi,

I tried but still getting an error:

```
ERROR: Nothing PROVIDES 'python' (but /home/build/workspace/recipes/python3-minimalmodbus/python3-minimalmodbus_git.bb DEPENDS on or otherwise requires it). Close matches:
  python3
  python3-py
  python3-sh
NOTE: Runtime target 'python3-minimalmodbus' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['python3-minimalmodbus', 'python']
ERROR: Required build target 'core-image-base' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-base', 'python3-minimalmodbus', 'python']
```

j123b567, 2021-11-27:
Did you try exactly what I wrote or you did some modifications?

In my recepie, there is no DEPENDS and also filename is different then yours.

j123b567, 2021-11-27:
It may be also necesery to exchange distutils3 by setuptools3

lemmi25, 2021-11-28:
Hi,

I had to change the checksum. This is my recipe (like you wrote):

```
SUMMARY = "Easy-to-use Modbus RTU and Modbus ASCII implementation for Python"
SECTION = "devel/python"
LICENSE = "Apache-2.0"
LIC_FILES_CHKSUM = "file://LICENCE;md5=27da4ba4e954f7f4ba8d1e08a2c756c4"

RDEPENDS_${PN} = "${PYTHON_PN}-pyserial"

SRCNAME = "minimalmodbus"
SRC_URI = "https://pypi.python.org/packages/source/m/${SRCNAME}/${SRCNAME}-${PV}.tar.gz"

SRC_URI[sha256sum] = "cf873a2530be3f4b86467c3e4d47b5f69fd345d47451baca4adbf59e2ac36d00"

S = "${WORKDIR}/${SRCNAME}-${PV}"

inherit distutils3
```

and it does not matter if I use distutils3 or setuptools3 the error is the same. Error msg:

```
ERROR: python3-minimalmodbus-2.0.1-r0 do_compile: 'python3 setup.py build ' execution failed.
ERROR: python3-minimalmodbus-2.0.1-r0 do_compile: Execution of '/home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/temp/run.do_compile.10383' failed with exit code 1
ERROR: Logfile of failure stored in: /home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/temp/log.do_compile.10383
Log data follows:
| DEBUG: Executing shell function do_compile
| /home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/recipe-sysroot-native/usr/bin/python3-native/python3: can't open file 'setup.py': [Errno 2] No such file or directory
| ERROR: 'python3 setup.py build ' execution failed.
| WARNING: /home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/temp/run.do_compile.10383:1 exit 1 from 'exit 1'
| ERROR: Execution of '/home/build/tmp/work/aarch64-poky-linux/python3-minimalmodbus/2.0.1-r0/temp/run.do_compile.10383' failed with exit code 1
ERROR: Task (/home/custom-layer/meta-custom-python/recipes-python/python/python3-minimalmodbus_2.0.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5190 tasks of which 5183 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/custom-layer/meta-custom-python/recipes-python/python/python3-minimalmodbus_2.0.1.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

j123b567, 2021-11-30:
It seems that the distutils3/setuptools3 is confused by `setup.cfg` which is used instead of `setup.py`. You will probably need to provide custom build step or ask Yocto community for help with this or add build time dependency to python3-build, or everything.

Sources contain `Makefile` which can be also a problem.

This is probably out of the scope of this issue tracker, but please come back when you solve so the documentation can be updated.

pyhys, 2022-02-17:
There is a tool to generate Yocto recipes for Python packages that are available on Pypi.
See https://github.com/NFJones/pipoe
I have not tried it my self, but it seems useful,

closed 2023-07-17 by pyhys