pyhys/minimalmodbus #81: [isar] Build minimalmodbus

MickaelDelanoe, 2021-11-30:
Hello,

I try to add minimalmodbus lib to my image with Isar Project.

So I do my recipe like follow
```
SUMMARY = "Easy-to-use Modbus RTU and Modbus ASCII implementation for Python."
DESCRIPTION = "MinimalModbus is an easy-to-use Python module for talking to \
instruments (slaves) from a computer (master) using the Modbus protocol, \
and is intended to be running on the master. Example code includes drivers \
for Eurotherm and Omega process controllers. The only dependence is the \
pySerial module (also pure Python). This software supports the 'Modbus RTU' \
and 'Modbus ASCII' serial communication versions of the protocol."

HOMEPAGE = "https://github.com/pyhys/minimalmodbus/"
SECTION = "devel/python"

SRC_URI="https://github.com/pyhys/minimalmodbus/archive/0.7.zip"
SRC_URI[md5sum] = "d40916acd676b6e94b9c8ce3e2f4e47c"

LICENSE = "Apache-2.0"
LIC_FILES_CHKSUM = "file://LICENSE;md5=27da4ba4e954f7f4ba8d1e08a2c756c4"

S = "${WORKDIR}/git"
DEPENDS = "python"
RDEPENDS_${PN} += "python-pyserial python-importlib"
#DEBIAN_DEPENDS = "python-importlib"

inherit dpkg

do_prepare_build[cleandirs] += "${S}/debian"

do_prepare_build() {
    deb_debianize
}
```

But when I build, I got the following error: 
```
Loading cache: 100% |#####################################################################################################################################################################| Time: 0:00:00
Loaded 59 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'python-pyserial' (but /repo/recipes-aquassay/minimalmodbus/minimalmodbus_0.7.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'python-pyserial' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['python-pyserial']
ERROR: Required build target 'iot2050-image-aquassay' has no buildable providers.
Missing or unbuildable dependency chain was: ['iot2050-image-aquassay', 'minimalmodbus', 'python-pyserial']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
2021-11-30 15:49:48 - ERROR    - Command returned non-zero exit status 1
```

Do you have an idea why `python-pyserial`is a trouble ?

Note that I my recipe is not in `ecipes-connectivity/minimalmodbus/` as write in documentation, I do not know if it can be a trouble ...

Best Regards,

j123b567, 2021-11-30:
I don't know what `Isar` is and I can't even find anything relevant. But from your recepie, it looks like some Buildroot/Yocto based system.

From the error message there is no way how to build `python-pyserial` because you have no package which provides it.
So just add recipe for pyserial.

It can also be similar problem to #79 so you should try e.g. `python3-pyserial` and update the overall dependency to `python` -> `python3`.

lemmi25, 2021-11-30:
Hi, 

this is the recipe I used for pyserial:

```
SUMMARY = "Python Serial Port Extension"
HOMEPAGE = "https://github.com/pyserial/pyserial"
AUTHOR = "Chris Liechti <cliechti@gmx.net>"
LICENSE = "BSD"
LIC_FILES_CHKSUM = "file://LICENSE.txt;md5=520e45e59fc2cf94aa53850f46b86436"

SRC_URI = "https://files.pythonhosted.org/packages/1e/7d/ae3f0a63f41e4d2f6cb66a5b57197850f919f59e558159a4dd3a818f5082/pyserial-3.5.tar.gz"
SRC_URI[md5sum] = "1cf25a76da59b530dbfc2cf99392dc83"
SRC_URI[sha256sum] = "3c77e014170dfffbd816e6ffc205e9842efb10be9f58ec16d3e8675b4925cddb

S = "${WORKDIR}/pyserial-3.5"

RDEPENDS_${PN} = ""

inherit setuptools3
``` 

This worked for me. Is your minimalmodbus recipe working after using my pyserial recipe?

j123b567, 2021-12-01:
If it is bitbake based distro, package name depends on bb filename. So, if the filename is e.g. `python3-serial_something.bb` then you should depend on `python3-serial` and not `python-pyserial`.

MickaelDelanoe, 2021-12-01:
Thanks for your replies.

> I don't know what Isar is and I can't even find anything relevant.

Well, firstable I built an image with Yocto, but to use my image on a new Hardware I have to move to Isar Project. I'm not an `Isar ` Expert, but basically, `Isar` is like an overlay of Yocto to help the build/dev/...

isar - Integration System for Automated Root filesystem generation
*(https://github.com/ilbers/isar)*

> because you have no package which provides it

Yes, I was confused because in my previous image I do not add recipe for pyserial. But I understand now, it's because I used `meta-openembedded` (who integrate pyserial) in my previous Yocto Project.

So yes, I will try to add pyserial recipe.
Thanks @lemmi25 for your recipe.

MickaelDelanoe, 2021-12-02:
Hi @lemmi25,
For now, I can't tel you if my minimalmodbus recipe work with your pyserial recipe because after build I got a new error

```
2021-12-02 10:10:52 - INFO     - /build$ /work/isar/bitbake/bin/bitbake -c build iot2050-image-aquassay
Loading cache: 100% |#####################################################################################################################################################################| Time: 0:00:00
Loaded 60 entries from dependency cache.
ERROR: ParseError at /repo/recipes-aquassay/python-pyserial/python-pyserial_3.0.1.bb:32: Could not inherit file classes/setuptools3.bbclass                                              | ETA:  --:--:--

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

So I added `setuptools3.bbclass`, but I got a new error

```
Loaded 60 entries from dependency cache.
Parsing recipes: 100% |###################################################################################################################################################################| Time: 0:00:00
Parsing of 60 .bb files complete (59 cached, 1 parsed). 60 targets, 0 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'python3-setuptools-native' (but /repo/recipes-aquassay/python-pyserial/python-pyserial_3.0.1.bb DEPENDS on or otherwise requires it)
ERROR: Required build target 'iot2050-image-aquassay' has no buildable providers.
Missing or unbuildable dependency chain was: ['iot2050-image-aquassay', 'python-pyserial', 'python3-setuptools-native']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
2021-12-02 08:19:58 - ERROR    - Command returned non-zero exit status 1
aquassay@debian-plasma:~/meta-iot2050$ 
```

And I don't know how provide `python3-setuptools-native` ...

I tried : IMAGE_PREINSTALL += "python3-setuptools" but not better ...

pyhys, 2022-02-17:
There is a tool to generate Yocto recipes for Python packages that are available on Pypi.
See https://github.com/NFJones/pipoe
I have not tried it my self, but it seems useful,

plpento, 2023-03-09:
Hi,

I am trying also to make yocto recipe.
I have tried pipoe but it is not working.
Did you find a solution?

Thanks,

MickaelDelanoe, 2023-04-19:
Hi,

Unfortunately no.
My project is in standby, I have no idea.
Sorry to can't help you.

Good luck,
-- 
Mickaël Delanoë
Ingénieur d'études | Design Engineer
T. +33 (0)5 87 03 80 59
E. ***@***.***
A. 31 rue du Châtenet 87410 Le Palais-sur-Vienne

https://aquassay.com/partnership-nestle-waters-aquassay/?lang=en

pyhys, 2023-07-17:
The documentation on Yocto is now updated: https://minimalmodbus.readthedocs.io/en/stable/advancedusage.html#including-minimalmodbus-in-a-yocto-build

closed 2023-07-17 by pyhys