pyhys/minimalmodbus #83: how to separate the debug mode information and the result ?

patfrench, 2022-01-19:
Hi

I have a tkinter application with 2 Text widgets, I would like to put in the first one the result and in the second one the debug mode details

Thank you

j123b567, 2022-01-19:
What do you mean by "result"?

patfrench, 2022-01-19:
Hi

widget 1 
```
MinimalModbus debug mode. Will write to instrument (expecting 13 bytes back): '\x01\x03\x00d\x00\x04\x05Ö' (01 03 00 64 00 04 05 D6)
MinimalModbus debug mode. Clearing serial buffers for port COM25
MinimalModbus debug mode. No sleep required before write. Time since previous read: 88594.00 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '\x01\x03\x08\x07\x0f\x00\x00\x00\x16\x01Ù\nÿ' (01 03 08 07 0F 00 00 00 16 01 D9 0A FF) (13 bytes), roundtrip time: 31.0 ms. Timeout for reading: 2500.0 ms.
```
widget2
```
1807
```

thx

j123b567, 2022-01-19:
`minimalmodbus` does not use standard logging and does not have hooks to register event listeners.
One solution I can imagine is that you can implement your own class based on the original and hook all functions you need like this

```python
class MyLoggingInstrument(minimalmodbus.Instrument):
    def _print_debug(self, text: str) -> None:
        do_something_with_debug_text(text)

    def read_long(self, *args, **kwargs):
        result = super().read_long(*args, **kwargs)
        do_something_with_read_long_result(result)
        return result

instrument = MyLoggingInstrument('/dev/ttyUSB1', 1)
```

There can be probably easier solution, but this should just work.

patfrench, 2022-01-21:
Hi

Thanks for the code, it will help me

Thank you

closed 2022-01-21 by patfrench