pyhys/minimalmodbus #84: minimalmodbus._unpack, why value = struct.unpack(formatstring, packed_bytes)[0] ?

patfrench, 2022-01-19:
Hi
I have to try :

```
minimalmodbus._unpack('>hhh', '\x07\x0f\x00\x00\x00\x16')
Out[69]: 1807

struct.unpack('>hhh',b'\x07\x0f\x00\x00\x00\x16')
Out[70]: (1807, 0, 22)
```

Why there is in source code, line 2605, index 0 ?
```
value = struct.unpack(formatstring, packed_bytes)[0]
```
Thank you

j123b567, 2022-01-19:
It is internal helper function to unpack one value. What are you trying to do and why are you using it directly?

patfrench, 2022-01-19:
I need to do a read_register**s** but with the signed option.
then i do 

```
minimalmodbus._unpack('>hhh', minimalmodbus._valuelist_to_bytestring(tete_ngt.read_registers(100,3,3), 3)
```

I need read a multiple registers int16, I hope I am clear ?

j123b567, 2022-01-20:
You have several possibilities

### Update source code of minimalmodbus
It should be easy to add support for parameter `signed` also for multiple registers read/write.
It seems for me, that it is rather artifical decision and everything is there to support it properly.
Then send a pull request.

So you can later do
```python
res = tete_ngt.read_registers(100, 3, 3, signed=True)  # not supported in current version
```

### Use two's complement
It is easy to implement two's complement function and convert value for only registers you really want.

```python
def twos_comp(val, bits):
    if val & (1 << (bits - 1)):
        val = val - (1 << bits)
    return val

# one register
res = tete_ngt.read_registers(100, 3, 3)
res[2] = twos_comp(res[2], 16)

# all registers
res = [twos_comp(r, 16) for r in tete_ngt.read_registers(100, 3, 3)]
```

### Use standard methods
Instead of using private functions, just use standard ones 

```python
import struct

def signed_registers(registers):
    repack = struct.pack("H"*len(registers), *registers)
    return struct.unpack("h"*len(registers), repack)

res = signed_registers(tete_ngt.read_registers(100, 3, 3))
```

patfrench, 2022-01-21:
Thanks for the examples. I will make a pull request.

closed 2022-11-24 by j123b567