pyhys/minimalmodbus #86: How to send a non-standard function code

larrybarello, 2022-05-07:
For a particular vendor, a custom function code ($78) with a payload of 0x0000, 0x0001, causes a reset to factory defaults.  The response appears to be an echo of the request.

Is this possible without modifying minimalmodbus code?  

i'm  a newbie and attempted a subclass, and ran into trouble pretty quick when making a stripped down "read" method that just jammed in the values.  

Could a generic write be exposed and allow the user jam in OEM functions?  Or maybe a magic method that overrides the function code on the next transaction?  Then the user can pick the closest transaction, override the function code.  In my case "read_register(0)" and poke in the $78  This is an ugly solution, but presumably the user would know what they are doing if they tried it.

pyhys, 2023-07-05:
Hi!
It is possible to use the _perform_command() functionality as described in 
https://minimalmodbus.readthedocs.io/en/stable/develop.html#extending-minimalmodbus

Note that it is outside the official API and is subject to future changes.

larrybarello, 2023-07-06:
Unfortunately, for my project I need to use 0xFF for the “broadcast” address and MinimalModbus only allows 0-247  For some reason known only to Renogy, they changed the broadcast address for their smart lithium batteries from 0 to 0xFF.   Thanks for the pointer to the dev documents.  Is there any reason I cannot just modify the code to allow 0xFF as an Broadcast adx?  I rummaged around the code a bit but was overwhelmed and dropped it.  I have more energy now and could revisit it again. Larry From: Jonas BergSent: Wednesday, July 5, 2023 10:19 AMTo: pyhys/minimalmodbusCc: larry Barello; AuthorSubject: Re: [pyhys/minimalmodbus] How to send a non-standard function code (Issue #86) Hi!It is possible to use the _perform_command() functionality as described inhttps://minimalmodbus.readthedocs.io/en/stable/develop.html#extending-minimalmodbusNote that it is outside the official API and is subject to future changes.