pyhys/minimalmodbus #88: Wrong functioncode: 3 instead of 16

AnonBit, 2022-10-19:
Read

    instrument = minimalmodbus.Instrument(port='COM1', slaveaddress=1, mode='rtu', debug=True)
    instrument.serial.baudrate = 4800  
    instrument.serial.bytesize = 8  
    instrument.serial.parity = serial.PARITY_NONE  
    instrument.serial.stopbits = 1   
    instrument.serial.timeout = 10 # seconds  
    instrument.clear_buffers_before_each_transaction = True  
      
    data = instrument.read_string(0)  
    print(data)

Write

    instrument = minimalmodbus.Instrument(port='COM3', slaveaddress=1, mode='rtu', debug=True) # port name, slave address (in decimal)  
    instrument.serial.baudrate = 4800  
    instrument.serial.bytesize = 8  
    instrument.serial.parity = serial.PARITY_NONE  
    instrument.serial.stopbits = 1  
    instrument.serial.timeout = 50  
    instrument.write_string(0, 'test')  
    instrument.serial.close()

Error

    minimalmodbus.InvalidResponseError: Wrong functioncode: 3 instead of 16. The response is: '\x01\x03\x00\x00\x00\x10D\x06'

j123b567, 2022-10-19:
Your report is not clear where the error happens. I suppose, that the part with "Read" example is completely irrelevant and the error happens in the `instrument.write_string()`

As you can see in the MODBUS specification. The response always repeats the slave address and function code from the request.
The request was something like `\0x01\0x10...` so the response must start with the same, which is not true indicating, that your slave device does not follow the MODBUS specification.

AnonBit, 2022-10-19:
Other example 
COM1(read_registers)  to COM2(write_registers)

I make request 

    # Will write to instrument (expecting 45 bytes back): 01 04 00 00 00 14 F0 05 (8 bytes)
    data = instrument.read_registers(0, 20, 4)

And I expect to response

    # 01 04 28 00 01 00 02 00 03 00 04 00 05 00 06 00 07 00 08 00 09 00 0A 00 0B 00 0C 00 0D 00 0E 00 0F 00 10 00 11 00 12 00 13 00 14 68 7A
    instrument.write_registers(0, [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20])

Error

    minimalmodbus.InvalidResponseError: Wrong functioncode: 4 instead of 16. The response is: '\x01\x04\x00\x00\x00\x14ð\x05

j123b567, 2022-10-19:
What are you trying to do? Running two modbus masters to communicate with each other? This will never work. Modbus was never designed to support this. Multiple masters on one bus can somehow tolerate/ignore themselves (highly nonstandard) but they will never be able to communicate with each other.

From the [README.rst](https://github.com/pyhys/minimalmodbus/blob/master/README.rst#features):
> MinimalModbus is an easy-to-use Python module for talking to instruments (slaves) from a computer (master) using the Modbus protocol, and is intended to be running on the **master**.

AnonBit, 2022-10-19:
Ok MinimalModbus can't as Slave

j123b567, 2022-10-19:
No, MinimalModbus is just master. You can use other libraries but implementing Modbus slave is not as easy as implementing master.

AnonBit, 2022-10-19:
Ok thanks

closed 2022-10-19 by AnonBit