pyhys/minimalmodbus #90: Bug in function "write_bit" with the value "1"

BAParticleCounter, 2022-11-21:
The code line is: instance.write_bit(16, 1 5)

The Exception:
System.ArgumentOutOfRangeException: The value must be greater than or equal to zero and less than the console's buffer size in that dimension.
Parameter name: top
Actual value was -1.
   at System.Console.SetCursorPosition(Int32 left, Int32 top)
   at Microsoft.PowerShell.PSConsoleReadLine.ReallyRender(RenderData renderData, String defaultColor)
   at Microsoft.PowerShell.PSConsoleReadLine.ForceRender()
   at Microsoft.PowerShell.PSConsoleReadLine.Insert(Char c)
   at Microsoft.PowerShell.PSConsoleReadLine.SelfInsert(Nullable`1 key, Object arg)
   at Microsoft.PowerShell.PSConsoleReadLine.ProcessOneKey(ConsoleKeyInfo key, Dictionary`2 
dispatchTable, Boolean ignoreIfNoAction, Object arg)
   at Microsoft.PowerShell.PSConsoleReadLine.InputLoop()
   at Microsoft.PowerShell.PSConsoleReadLine.ReadLine(Runspace runspace, EngineIntrinsics engineIntrinsics)

If I use this function with the value of "0", it is working properly!

j123b567, 2022-11-21:
Your example is incomplete and syntactically wrong, but the problem is probably in your own code that you didn't share.

 -  `Microsoft.PowerShell` is clearly not a python package
 - exception is clearly not in `minimalmodbus`

BAParticleCounter, 2022-11-21:
So this is my code. Everything is working. I can read registers. I can stop the working device. But I can't start it. 

`import minimalmodbus
from datetime import datetime
from time import sleep

deviceAddress = 1
CONST_FLOWRATE = 28.3
inhalt = {}

particle_counter = minimalmodbus.Instrument('COM3', deviceAddress, minimalmodbus.MODE_RTU, debug=False) 

particle_counter.serial.baudrate = 9600				    # BaudRate
particle_counter.serial.bytesize = 8					# Number of data bits to be requested
particle_counter.serial.parity = minimalmodbus.serial.PARITY_NONE	# Parity Setting here is NONE but can be ODD or EVEN
particle_counter.serial.stopbits = 1					# Number of stop bits
particle_counter.serial.timeout  = 0.1      			# Timeout time in seconds

particle_counter.clear_buffers_before_each_transaction = True
particle_counter.close_port_after_each_call = True

particle_counter.write_bit(16, 1, 5)

hexParticleCount300nm   = particle_counter.read_registers(3, 2, 4)
hexParticleCount500nm   = particle_counter.read_registers(5, 2, 4)
hexParticleCount1000nm  = particle_counter.read_registers(7, 2, 4)
hexParticleCount2500nm  = particle_counter.read_registers(9, 2, 4)
hexParticleCount5000nm  = particle_counter.read_registers(11, 2, 4)
hexParticleCount10000nm = particle_counter.read_registers(13, 2, 4)

decParticleCount300nm   = (hexParticleCount300nm[0]   +hexParticleCount300nm[1])   /CONST_FLOWRATE
decParticleCount500nm   = (hexParticleCount500nm[0]   +hexParticleCount500nm[1])   /CONST_FLOWRATE
decParticleCount1000nm  = (hexParticleCount1000nm[0]  +hexParticleCount1000nm[1])  /CONST_FLOWRATE
decParticleCount2500nm  = (hexParticleCount2500nm[0]  +hexParticleCount2500nm[1])  /CONST_FLOWRATE
decParticleCount5000nm  = (hexParticleCount5000nm[0]  +hexParticleCount5000nm[1])  /CONST_FLOWRATE
decParticleCount10000nm = (hexParticleCount10000nm[0] +hexParticleCount10000nm[1]) /CONST_FLOWRATE

inhalt = {
    'Timestmp': datetime.now().strftime("%d/%m/%Y %H:%M:%S"),
    '300nm' :   round(decParticleCount300nm),
    '500nm' :   round(decParticleCount500nm), 
    '1000nm' :  round(decParticleCount1000nm),
    '2500nm' :  round(decParticleCount2500nm),
    '5000nm' :  round(decParticleCount5000nm),
    '10000nm':  round(decParticleCount10000nm)
}
i=+1
particle_counter.write_bit(16, 0, 5)`

j123b567, 2022-11-21:
This code looks ok (FYI you can use triple ``` to preserve formatting in a code block).

So other questions...

What is your python interpreter? **Version** and type - regular Python (CPython), IronPython, PyPy, ...

What is the minimalmodbus version or how do you obtain it? By `pip install minimalmodbus`, from git, ...

Can you attach the complete output of the script you have provided as an example? Probably just some exception, it does not print anything...

BAParticleCounter, 2022-11-21:
- Regular Python V3.7.8

- Over "py -m pip install minimalmodbus" - It is the Version 2.0.1

The script is working. So it don't interrupt. But sometimes (not all the time) i get this output besides the path where i start the script from: 

"Oops, something went wrong.  Please report this bug with the details below.
Report on GitHub: https://github.com/lzybkr/PSReadLine/issues/new
-----------------------------------------------------------------------
Last 200 Keys:
 o d b u s M a s t e r P y t h o n / m o d b u s t e s t c o d e . p y Enter
 & Space " C : / P r o g r a m Space F i l e s Space ( x 8 6 ) / M i c r o s o f t Space 
V i s u a l Space S t u d i o / S h a r e d / P y t h o n 3 7 _ 6 4 / p y t h o n . e x e " Space c : / U s e r s / u s e r 0 1/ D o c u m e n t s / 0 0 _ T e s t o r d n e r M o d b u s / 0 6 _ S o f t w a r e / M o d b u s M a s t e r P y t h o n / t e s  . p y Enter 

Exception:
System.ArgumentOutOfRangeException: The value must be greater than or equal to zero and less than the console's buffer size in that dimension.
Parameter name: top
Actual value was -2.
   at System.Console.SetCursorPosition(Int32 left, Int32 top)
   at Microsoft.PowerShell.PSConsoleReadLine.ReallyRender(RenderData renderData, String defaultColor)
   at Microsoft.PowerShell.PSConsoleReadLine.ForceRender()
   at Microsoft.PowerShell.PSConsoleReadLine.Insert(Char c)
   at Microsoft.PowerShell.PSConsoleReadLine.SelfInsert(Nullable`1 key, Object arg)      
   at Microsoft.PowerShell.PSConsoleReadLine.ProcessOneKey(ConsoleKeyInfo key, Dictionary`2 dispatchTable, Boolean ignoreIfNoAction, Object arg)
   at Microsoft.PowerShell.PSConsoleReadLine.InputLoop()
   at Microsoft.PowerShell.PSConsoleReadLine.ReadLine(Runspace runspace, EngineIntrinsics engineIntrinsics)

BAParticleCounter, 2022-11-23:
Oh wow why didn't i think of setting debug = true...Beginner's mistake haha

I have only taken the important output:

MinimalModbus debug mode. Will write to instrument (expecting 8 bytes back): 01 05 00 10 FF 
00 8D FF (8 bytes)
MinimalModbus debug mode. Clearing serial buffers for port COM3
MinimalModbus debug mode. No sleep required before write. Time since previous read: 91786703.00 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Closing port COM3
MinimalModbus debug mode. Response from instrument: 01 05 00 10 FF 00 8D FF (8 bytes), roundtrip time: 0.0 ms. Timeout for reading: 100.0 ms.

So the way I see it. The **FF 00** should be **00 01**.

BAParticleCounter, 2022-11-24:
Problem solved!

In the "minimalmodbus.py" file,  beginning at line 2766:
```
if value == 0:
        return "\x00\x00"
    return "\xff\x00"
```

For my application I need the hexcode **00 01** and not **ff 00**. 

So I changed it and now it is working. 

Thank you for your help @j123b567 !

closed 2022-11-24 by BAParticleCounter