pyhys/minimalmodbus #91: InvalidResponseError: Wrong functioncode

ollie-campbell, 2022-11-22:
Hi,
I'm using minimalmodbus (2.0.2) to read data from a system connected to a Raspberry Pi over USB-to-serial convertor and it's working well. I would like to try and write data to it now, but am running into a problem.

When I read a register I get the following:

> >>> instrument.read_register(192)
> MinimalModbus debug mode. Will write to instrument (expecting 7 bytes back): C9 03 00 C0 00 01 94 7E (8 bytes)
> MinimalModbus debug mode. Clearing serial buffers for port /dev/serial0
> MinimalModbus debug mode. No sleep required before write. Time since previous read: 127454.06 ms, minimum silent period: 4.01 ms.
> MinimalModbus debug mode. Response from instrument: C9 03 02 00 04 58 57 (7 bytes), roundtrip time: 0.1 ms. Timeout for reading: 150.0 ms.
> 
> 4

If I try to write to this register I get the following:

> >>> instrument.write_register(192, 0)
> MinimalModbus debug mode. Will write to instrument (expecting 8 bytes back): C9 10 00 C0 00 01 02 00 00 80 95 (11 bytes)
> MinimalModbus debug mode. Clearing serial buffers for port /dev/serial0
> MinimalModbus debug mode. No sleep required before write. Time since previous read: 7018.76 ms, minimum silent period: 4.01 ms.
> MinimalModbus debug mode. Response from instrument: C9 01 00 A0 6E (5 bytes), roundtrip time: 0.2 ms. Timeout for reading: 150.0 ms.
> 
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/home/pi/.local/lib/python3.9/site-packages/minimalmodbus.py", line 550, in write_register
>   File "/home/pi/.local/lib/python3.9/site-packages/minimalmodbus.py", line 1245, in _generic_command
>     payload_from_slave = self._perform_command(functioncode, payload_to_slave)
>   File "/home/pi/.local/lib/python3.9/site-packages/minimalmodbus.py", line 1329, in _perform_command
>     payload_from_slave = _extract_payload(
>   File "/home/pi/.local/lib/python3.9/site-packages/minimalmodbus.py", line 1885, in _extract_payload
>     raise InvalidResponseError(
> minimalmodbus.InvalidResponseError: Wrong functioncode: 1 instead of 16. The response is: 'É\x01\x00\xa0n'

Am I reading it correctly that the unit I am querying is responding from a different function code and minimalmodbus doesn't like it?

ollie-campbell, 2022-11-22:
Apologies 2.0.1, a mistype.

Ok thank you. I realise this isn't a bug issue now, but are you able to point me in a diagnosis direction?

j123b567, 2022-11-23:
Maybe the Modbus protocol specification https://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf

Chapter **6.12 16 (0x10) Write Multiple registers**

ollie-campbell, 2022-11-23:
Thank you for your help so far. I have had a response from the manufacturers, who have given me Hex and binary to write to the device.
Can I use this directly using minimalmodbus? Or is there a way I can 'convert' this to use it?

Examples from them are:

201 6 10753 78
C9 06 2A 01 00 4E 406E

Thanks again.

j123b567, 2022-11-23:
In your example, the function used is **06 Write Single Register**
 - You can try `instrument.write_register(10753, 78, functioncode=6)`
 - You can simply try to patch minimalmodbus in your project to ignore invalid responses generated by your instrument.

ollie-campbell, 2022-11-24:
Great thank you once again.

closed 2022-11-24 by ollie-campbell