pyhys/minimalmodbus #93: Why cannot you use hex numbers as registers?

bmoneke, 2022-11-24:
The docstrings state for the registeraddress: "use decimal numbers, not hex". However, you can give an integer like `0x10` and it will be an integer of value `16`, as far as I know.

j123b567, 2022-11-24:
I'm not the author, but there may be a lot of documentation on some instrument that use numbers in base 16, but without suffixes and prefixes. So basically it can mean "watch the base of the number".

So the wording should be probably better and you can freely use `0x10` instead of `16`. But if you see `10` in the documentation of some instrument, you should be careful if they mean `10` or `0x10`.

bmoneke, 2022-11-24:
Thanks for the clarification.

closed 2023-02-01 by bmoneke