pyhys/minimalmodbus #96: No answer

Grandpa-G, 2023-02-01:
I am getting this error on both read and writes periodically, but almost always. Using a Rpi4 bullseye. I know the device works fine on a windows application. Where do I go from here?

  ```
  self.box.write_register(registeraddress=switchChange-1, value=1, number_of_decimals= 0, functioncode= 16, signed= False)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 550, in write_register
    self.box.write_register(registeraddress=switchChange-1, value=1, number_of_decimals= 0, functioncode= 16, signed= False)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 550, in write_register
    self._generic_command(
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1245, in _generic_command
    self._generic_command(
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1245, in _generic_command
    payload_from_slave = self._perform_command(functioncode, payload_to_slave)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1322, in _perform_command
    payload_from_slave = self._perform_command(functioncode, payload_to_slave)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1322, in _perform_command
    response_bytes = self._communicate(request_bytes, number_of_bytes_to_read)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1490, in _communicate
    response_bytes = self._communicate(request_bytes, number_of_bytes_to_read)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1490, in _communicate
    raise NoResponseError("No communication with the instrument (no answer)")
minimalmodbus.NoResponseError: No communication with the instrument (no answer)
    raise NoResponseError("No communication with the instrument (no answer)")
minimalmodbus.NoResponseError: No communication with the instrument (no answer)
```

Grandpa-G, 2023-02-02:
Thanks, but this happens almost every time after the first success. I am using the same cables as windows. I done the debug, and that is what I got as I posted. Ignoring is not an option.

So far I have moved to pyModBus and it is working.

pyhys, 2023-07-04:
Closing due to inactivity

closed 2023-07-04 by pyhys