pyhys/minimalmodbus #101: Modbus ID change - Wrong return slave address

somakelemen, 2023-05-05:
Hi,

I have a small script which change the slave default Modbus ID to another one (from 1 to 23 or any other unused ID) via write_register().
My issue is when it's done I got the following "error" which is right but basically this was the goal.

`minimalmodbus.InvalidResponseError: Wrong return slave address: 23 instead of 1. The response is: '\x17\x06@\x03\x00\x17.ò'`

I am using,
Python 3.9.2
minimalmodbus 2.0.1

Are there any option to override the expected slave address?

pyhys, 2023-07-04:
Maybe you can just silence the error for that particular write:
```
try:
    instrument.write_register(24, NEW_TEMPERATURE, 1) 
except minimalmodbus.InvalidResponseError:
    pass
```

penguintamer, 2024-03-30:
Another time disabling return slave address checking would be useful: accessing single connected devices by their "universal" address such as 0xF7 or 0xFF.

Unfortunately, silencing errors with a try block has the unfortunate side effect of making the value returned from the Instrument unusable except by trying to parse the string that comes back in the error. So a simple pattern like using the universal address to query the register that contains the device's real address won't work.

Janl1, 2024-04-01:
In my case I wanted to change the slave address of an Epever XTRA3210N Solar Charge Controller.

With some reverse engineering of the official Epever software I found out that you cannot change the slave address to a normal register value, but you can use the function code 69 to change it.

I used the following line of code to change the address. It will probably return an error message because it cannot parse the result, but the change went through in my case.

`instrument._perform_command(69, instrument._num_to_onebyte_string(newId))` (minimalmodbus v2.0.1)

Please keep in mind that the function code might be different depending on the device and that playing around with undocumented function codes can break the communication with the device.