pyhys/minimalmodbus #127: If you read the data multiple times, you won't be able to read the data

childeyouyu, 2024-01-31:
```
    while True:
        read_data = inst.read_float(registeraddress=1, number_of_registers=2, functioncode=3)
        print(read_data)
        time.sleep(1)
```
I need to consistently read the data in the device, but frequently, it will not read the data.
Like this:
minimalmodbus.NoResponseError: No communication with the instrument (no answer)

arthurtaras, 2024-01-31:
Increase the sleep time to 50 and see if errors still appear. If not decrease to 25, 10, 5 and so on. Chances are you are overdriving it too fast either on the processor or microcontroller side

childeyouyu, 2024-02-01:
@arthurtaras Do you mean 50s? This is too long a wait time for me, and I need to record a lot of data on demand.

j123b567, 2024-02-01:
Did you follow all instructions in [Serial communication][1] and [Troubleshooting][2]?

[1]: https://minimalmodbus.readthedocs.io/en/stable/serialcommunication.html
[2]: https://minimalmodbus.readthedocs.io/en/stable/troubleshooting.html

arthurtaras, 2024-02-01:
@childeyouyu just test it with 10, 5, 2 and see if the frequency of errors goes up or down. Do you have access to the code on the microcontroller? To read more data more frequently you would need to optimize the other side

childeyouyu, 2024-02-01:
> @childeyouyu just test it with 10, 5, 2 and see if the frequency of errors goes up or down. Do you have access to the code on the microcontroller? To read more data more frequently you would need to optimize the other side

I  will try it tomorrow.

childeyouyu, 2024-02-02:
@j123b567 
It works, I set `inst.serial.timeout` to 1, the program reads a thousand lines in a row without the previous error, I will test the program all the way to the official program code, and if available, I will close this issue.
So is this due to a problem with my device's performance?

closed 2024-02-02 by childeyouyu

j123b567, 2024-02-02:
@childeyouyu There is nothing particularly wrong with that. Some devices even add a programmable delay before responding. You can try a much smaller number than 1 second, e.g. `0.03` or even `0.01` might do. Setting this to 1 can make the whole communication unnecessarily slow.