pyhys/minimalmodbus #129: If I want to read different device addresses in the same serial port, can I do it?

childeyouyu, 2024-02-23:
The code is as follows：
```
    for i in range(1, 6 * 2, 2):
        try:
            inst = minimalmodbus.Instrument('COM3', 1)
            inst.serial.baudrate = 9600
            inst.serial.parity = "N"
            inst.serial.bytesize = 8
            inst.serial.stopbits = 1
            inst.close_port_after_each_call = True
            inst.serial.timeout = 1
            temp = inst.read_float(registeraddress=i, number_of_registers=2, functioncode=3)
            print(temp)
        except Exception as e:
            print(i, e)
            
    for i in range(2, 6 * 2, 2):
        try:
            inst = minimalmodbus.Instrument('COM3', 2)
            inst.serial.baudrate = 9600
            inst.serial.parity = "N"
            inst.serial.bytesize = 8
            inst.serial.stopbits = 1
            inst.close_port_after_each_call = True
            inst.serial.timeout = 1
            temp = inst.read_float(registeraddress=i, number_of_registers=2, functioncode=3)
        except Exception as e:
            print(i, e)
```
Since I don't have a device at hand that can perform this operation, but the demand is to perform this function, I would like to get some answers.

j123b567, 2024-02-23:
Yes https://minimalmodbus.readthedocs.io/en/stable/usage.html#using-multiple-instruments

childeyouyu, 2024-02-23:
> Yes https://minimalmodbus.readthedocs.io/en/stable/usage.html#using-multiple-instruments

thanks