pyhys/minimalmodbus #137: RS485 Transceiver Chip

CrabbyPete, 2025-02-26:
I am using minimalmodbus on a Raspberry Pi that has an add on board with a RS485 transceiver chip that required me to set the DE pin to control writing and reading from the chip. I can do it manually like this 

```
def write_485():
    s=serial.rs485.RS485(port='/dev/serial0', baudrate=9600, parity=serial.PARITY_NONE)
    s.rs485_mode = serial.rs485.RS485Settings(False,True)
    s.write('ABC'.encode('utf-8'))
    s.flush()
    GPIO.output(TXDEN, GPIO.LOW)
    wait_for = 5
    while wait_for:
        if s.in_waiting > 0:
            data = s.read(s.in_waiting)
            print(f"Received: {data}")
            log.info(f"Recieved:{s.in_waiting}")
            wait_for -= 1
```
Is there a way to do it in the minimalmodbus module or do I need to re-write it?

j123b567, 2025-02-26:
It is generally a bad idea to handle DE manually in Python, because it is slow and unreliable.
That said, what about this?

```python
class MyRS485(serial.rs485.RS485):
    def setRTS(self, value):
        GPIO.output(TXDEN, GPIO.HIGH if value else GPIO.LOW)

ser  = MyRS485(port='/dev/serial0', baudrate=9600, parity=serial.PARITY_NONE)
ser.rs485_mode = serial.rs485.RS485Settings(False,True)
instrument = minimalmodbus.Instrument(ser, 1)
```

CrabbyPete, 2025-02-26:
Thanks, I 'll try it. I know its not ideal, but its a 9600 baud interface that is very simple

CrabbyPete, 2025-02-28:
Works great, thanks. Only thing is the logic seems to be reversed

`    def setRTS(self, value):
        GPIO.output(TXDEN, GPIO.LOW if value else GPIO.HIGH)
`

j123b567, 2025-02-28:
It depends on your hardware and on `RS485Settings(False, True)` vs `RS485Settings(True, False)`

closed 2025-03-07 by j123b567